opentx/opentx #3362: Feature Request Activity Log

jrumbelow, 2016-03-14:
Hi Guys. Dont know if this is the right place to request this, First off all big thank you to you all for the development of open tx. I would like to request an activity log like this http://blog.open-tx.org/2016/03/02/new-popups-alerts-style/  on the taranis. Ill explain the situation i have experienced. So flying on slope cant hear the speaker clearly when its super windy, you then look down at screen and go through your telemetry to see whats happening which can be a bit hit and miss if the issue has gone away, for example battery voltage on plane has dipped below a certain value. How about another screen on the telemetry or popup that shows in big text the last recorded major incident like battery or rssi. What do you guys think, is the taranis capable and once again a big thank you for making my taranis what is is. Its super!

kilrah, 2016-03-15:
That would be a job for a lua script IMO.

projectkk2glider, 2016-03-15:
Exactly. And what is considered as a exceptional condition is also very dependent on the particular model/setup. So making anything generic would probably not work as desired (except maybe for RSSI).

jrumbelow, 2016-03-15:
My main concern when flying is battery voltage on the plane and rssi. What about everything you set in custom switches. Have a check box to select whether you want a visual alarm too. Is there any resources regarding lua scripts that might help me to have ago at this.  Kind regards Jordan

projectkk2glider, 2016-03-15:
Start on forums, for example http://www.rcgroups.com/forums/showthread.php?t=2180477

closed 2019-11-15 by stale[bot]