opentx/opentx #3365: google earth not opening in 2.1.7

nigelsheffield, 2016-03-17:
I cant get get anything to happen when I try to link to google earth, I have set the path as I did in 2.0 which worked but now nothing happens when I press the google earth button.
I have not changed any names of any sensors.
windows 7 taranis none plus.

projectkk2glider, 2016-03-18:
Do you get any dialog message with an error?

projectkk2glider, 2016-03-18:
Please upload a log file example somewhere and post link here, so I can do a test here.

nigelsheffield, 2016-03-18:
http://www.rcgroups.com/forums/showpost.php?p=34290206&postcount=34

a link to the file which I cant get to open to google earth. No error message just nothing happens, if I select an old file from 2.0 it gives an error message about gps wrong labels.

kilrah, 2016-03-18:
No issue here.

[image]

nigelsheffield, 2016-03-18:
I must have done something wrong then, because i still get no response from this file, i still have 2.0 installed and that also does nothing also with this file but does work with old files from 2.0 
but 2.1 reports a problem with old files and just does nothing with this file...

nigelsheffield, 2016-03-18:
ok i got it linking now , i had wrong executable selected sorry.
I dont see any data on screen though, I think i have setting wrong for altitude or something in google earth?

nigelsheffield, 2016-03-18:
I can get to see the path by adjusting height but altitude tracking looks wrong, all the same level to look  at though when clicking on parts of path it says different altitudes.

closed 2016-03-18 by nigelsheffield

projectkk2glider, 2016-03-18:
Try different altitude setting in Google Earth track properties, like clamp to ground, etc...

Did you close ths by accident?

nigelsheffield, 2016-03-18:
I tried different setting  under edit , properties and tried every track option for altitude , maybe I need to enable GPS altitude logging for it to work but I did have baro altitude logged.
To be honest I don't know what to expect as this is the first time I've had GPS and don't know whether the map tracking graph tracks the altitude or if it is just displayed in a graph.
I closed it because it's obviously nothing wrong with companion and just my lack of knowledge ...

kilrah, 2016-03-18:
> maybe I need to enable GPS altitude logging for it to work but I did have baro altitude logged.

Yes at this point the Google Earth export only looks at the "GAlt" column. 

Making the GPS/Alt/Speed fields selectable is a possible improvement listed in #2569.