opentx/opentx #3366: Log File Creation, V2.1.7, Taranis

flyandhike, 2016-03-18:
Hello
Log Files of the Telemetry data will be created one per day and per Model Type. When the Log parameters will be changed during the day, ie. add a new sensor in the telemetry page, the Log File gets corrupted. You will see this, when you would like to show such a Log File on the Companion graph page: an error message shows up ...invalid lines.... The Log does save all parameters, but it's missing the new equivalent header line. Therefore it gets a mismatch and does not know, what to do with the added parameter column.
By knowing this, delete or rename the old Log File of the day, before starting a new Log. In future, may you correct this bug by automatically insert a new header line or create a new Log File?

kilrah, 2016-03-18:
Duplicate of #2542

closed 2016-03-18 by kilrah

flyandhike, 2016-03-18:
Thank you Andre for your reply.

 

Closed #3366 <https://github.com/opentx/opentx/issues/3366> .