opentx/opentx #3370: Horurs consumption measurement

projectkk2glider, 2016-03-18:
This is not actual issue, but a data store.

Horus consumption on commit eba944f55b1cbb1f7d3ee038c8d571c72c7d2b22:
* radio on, back-light off, TX off = 150 - 160 mA
* radio on, back-light on, TX off = 195 - 205 mA
* TX in D16 or D8 mode = 245 - 255 mA
* TX in LR12 mode = 275 mA
* radio off = 16uA

projectkk2glider, 2016-03-23:
Component consumption (estimated from data-sheets and measured):
* back-light: 40mA @10V
* IXJT: 50 - 70mA @10V
* CPU 100mA @3.3V -> 40mA @10V (80% power supply efficiency)
* MP3 40 - 75mA @3.3V
* Audio amp: up to 260mA @5V
* LCD: unknown
* GPS: unknown
* ADC: 2mA
* SDRAM: 2 - 150mA @3.3V

So there is not much sense in trying to reduce CPU consumption (by using sleep mode in idle task). Max we could gain is 30mA at battery which is only 8% improvement.

mac44mag, 2016-05-25:
I have a battery analyzer, and will attempt to run a set of charge/discharge curves in the next few days.  Will get back to you.

Leonard

closed 2016-08-01 by bsongis