opentx/opentx #3372: companion 2.1.7

nostromo123, 2016-03-20:
Just found an issue in companion 2.1.7
Telemetry screen in companion only offers 2 choices. See attached screen shot. 
FrSky D (cable) is missing in companion. Needed for our MPX FrSky telemetry converter to get the data into Taranis ...
[image]

kilrah, 2016-03-20:
Indeed. You can set it on the radio itself in the meantime.

nostromo123, 2016-03-20:
Took me half an hour to find out ;-)
Was looking for any issues with our converter
br KH

nostromo123, 2016-03-20:
Sorry, not used to the buttons here. Was my #1 at github ...

projectkk2glider, 2016-03-20:
Don't worry, it often happens. :smile:

commit 13d5fe56 (2016-04-28): Fixes #3372: Telemetry mode "FrSky D (cable)" choice was missing for Taranis
commit 92c76b58 (2016-04-28): Fixes #3372: Telemetry mode "FrSky D (cable)" choice was missing for Taranis (ported from master)
closed 2016-04-28 by projectkk2glider