opentx/opentx #3375: Volume adjust non-linearity

projectkk2glider, 2016-03-20:
When using pot for volume adjusting a value of -50% is almost silent. There is not enough precision at low volume levels. It behaves like not enough exponent is used.

kilrah, 2016-03-20:
Only on Horus?`Could be that the audio chip has log volume control while it was linear on other platforms.

bsongis, 2016-03-20:
It is not linear on other platforms, there is an adaptation table. I don't remember, perhaps I removed it on Horus and we need it again.

projectkk2glider, 2016-03-20:
The MP3 player takes the volume in decibels, so there is no need to have an adaptation table in firmware! I will try to fix this.

bsongis, 2016-03-20:
```
const uint8_t volumeScale[VOLUME_LEVEL_MAX+1] = {
  254, 127, 110, 100, 90, 80, 70, 60, 50, 40,
  30, 20, 19, 18, 17, 16, 15, 14, 13, 12,
  11, 10, 5, 0
};
```

projectkk2glider, 2016-03-20:
:arrow_up: I think the steps should be equidistant - ie a simple linear function. I am trying my fix now..

commit 28dd7190 (2016-03-20): Fixes #3375: better linearity for Horus volume control
closed 2016-03-20 by projectkk2glider