opentx/opentx #3376: companion 2.1.7 - German text showing up in English GUI

nostromo123, 2016-03-21:
Only a minor issue, but stumbled over those German words showing up in English companion GUI?
Have both opentx and companion set to English language ....
[image]

kilrah, 2016-03-21:
Looks like your system language is set to German, and automatically replaces some standard strings?

Can't reproduce here.
Are you running the officially distributed package or a self-compile?

nostromo123, 2016-03-21:
Official package, system is Debian Jessie KDE 64 bit.
Linux system language is set to German.

nostromo123, 2016-03-21:
Strange.
Have a backup Debian Jessie KDE 64 bit in a vbox. Set that system to "system language" English.
Installed companion ... No issue there .... Don't understand, what causes that language mixture????

[image]

kilrah, 2016-03-21:
It must be something on your system, given that nowhere in the OpenTX code do either of the 2 words appear, not even in the German translation files!

closed 2016-03-21 by kilrah