opentx/opentx #3377: make call to background alternate with call to run in lua script between cycles

lshems, 2016-03-21:
In openTX 2.0 we had to call the background script from the run script if we still wanted it to execute during run. This implies that you had only one processing cycle (limited to 25 ms??) if you wanted to execute both.

In 2.1, the background is always called before the run. This opens up the possibility to assign the calculation cycles alternating between background and run, to have 25 ms available to both. 

????

kilrah, 2016-03-22:
Is it an observation or a question? If the latter... what's the question?

lshems, 2016-03-22:
Question: is my observation correct that in 2.1 the timeslice for lua telemetry scripts is used for both the background and the run, or do they have a separate timeslice?

If the first, I'm not that happy, and would request giving both an individual timeslice, as time is critical already.

projectkk2glider, 2016-11-30:
When you are talking about `timeslice` I presume you are talking about the Lua script limits - ie how many instructions can be executed. If yes, then each periodic invocation of any script function starts counting instructions from zero. 

So in case of the telemetry script both `background` and `run` each get their own timeslice. 

The point of limiting the Lua script is to kill scripts that are taking too long. The `25ms` is an arbitrary number, the actual limit is the number of Lua internal instructions. So the limit is not in time, but instructions. 

The whole idea is not to block the main thread (that is responsible for graphics and keys) for too long time. If your scripts are getting killed because of this you should consider doing less in each invocation - you should split the work between multiple script invocations.

lshems, 2016-12-01:
Ok, then we can close this. Thanks for the clarification.

closed 2016-12-01 by lshems