opentx/opentx #3380: Support for Light-telemetry protocol

marksev1, 2016-03-22:
Its a low baudrate protocol awesome for transmitting data around with veryvery low effect on range of transmission vs the more heavier protocols.

It is implemented already in Cleanflight, Inavflight probably also Betaflight and in Taulabs. 

Use cases for LTM-protocol:

Ghettostation and open360tracker antenna tracker and telemetry display
Ghettosd and ltm-osd (either ground or air OSD)
Mwptools software supports ltm (it allows logging, has audio prompts, displays data, map etc.)
I have all of those above ready for testing :)
Also EZ-gui support it

Would be really nice if this project would support it natively.

kilrah, 2016-03-22:
How do you intend to input it to the radio?

marksev1, 2016-03-22:
Lol open360tracker&open360tracker-amv is a hugely popular antenna tracker just google it on fpv-community.de, also the diy HD-fpv osd from Rangarid uses it. And I'm sure you know all the above multicopter projects that I mentioned. Also I'm sure you are familiar with EZ-gui the android app for configuring the various multicopter projects?

I'd input it via serial connection, for example via transparent serial link like openlrsng does it, dunno...I'd have to study a bit regarding the hardware. Like how do guys that have mavlink on their transmitters input it, i'd do it the same way. 

Implementation in Taulabs: https://github.com/TauLabs/TauLabs/blob/00c034abddff45b1fd033b23e062e210d50c542b/flight/Modules/UAVOLighttelemetryBridge/UAVOLighttelemetryBridge.c
Implementation in Inav: https://github.com/iNavFlight/inav/blob/master/src/main/telemetry/ltm.c

Here's how KipK the one who wrote the first draft of the protocol described it (I don't know if it changed any by now but I think not):

LTM is an air to ground telemetry, so it's one way only; No ACK & stuffs it just stream datas to the ground at low baudrates.

Depending of flighcontroller implementation & configuration it starts at boot or when armed.

Protocol is binary, using little endian.
There's 3 frames to consider:

G Frame (GPS position) (2hz @ 1200 bauds , 5hz >= 2400 bauds): 18BYTES 0x24 0x54 0x47 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xC0
$ T G --------LAT-------- -------LON--------- SPD --------ALT-------- SAT/FIX CRC
A Frame (Attitude) (5hz @ 1200bauds , 10hz >= 2400bauds): 10BYTES
0x24 0x54 0x41 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xC0

$ T A --PITCH-- --ROLL--- -HEADING- CRC

S Frame (Sensors) (2hz @ 1200bauds, 5hz >= 2400bauds): 11BYTES
0x24 0x54 0x53 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xC0

$ T S VBAT(mv) Current(ma) RSSI AIRSPEED ARM/FS/FMOD CRC

values are:

LAT/LON: int32 (cm)
SPD ( speed ) : uint8 ( m/s )
ALT ( alititude ) int32 ( cm )
SATS ( number of sat visible ) 6bits
FIX ( sat fix ) 2 bits ( sats & fix are coupled in the same byte )
PITCH/ROLL/HEADING int16 ( deg , -180/180° range )
VBAT( voltage ) uint16 (mv )
Current uint16 ( ma )
RSSI : uint8 ( in % )
AIRSPEED : uint8 (m/s)
ARM/FS/FMOD: uint8 ARM armed status: 1 bit ( first ) FS : failsafe status : 1 bit ( second one ) FMOD: flight mode : last 6 bits: // Flight mode(0-19): 0: Manual, 1: Rate, 2: Attitude/Angle, 3: Horizon, 4: Acro, 5: Stabilized1, 6: Stabilized2, 7: Stabilized3, // 8: Altitude Hold, 9: Loiter/GPS Hold, 10: Auto/Waypoints, 11: Heading Hold / headFree, // 12: Circle, 13: RTH, 14: FollowMe, 15: LAND, 16:FlybyWireA, 17: FlybywireB, 18: Cruise, 19: Unknown
Each frame end with a CRC byte, it's a simple XOR from first payload byte to last one ( starting at 4th byte , headers are not xored )

marksev1, 2016-04-01:
Any feedback or should I close the issue?

kilrah, 2016-04-01:
You can leave it open should someone be interested in doing the work, but it likely won't be the current team. It seems very redundant with the FrSky smart port protocol, especially given that a number of flight controllers already support it natively.

sppnk, 2016-04-27:
I am very interested, because Frsky implementation has no error control, and a one-way light protocol like LTM (with chksum) is a very good solution.

bsongis, 2016-04-27:
FrSky implementation has no error control. Are you sure?

sppnk, 2016-04-27:
AFAIK there is no control of data sent. No checksum:

[http://www.frsky-rc.com/download/down.php?id=126](url)

In my Turnigy 9x the GPS data or altitude data are not stable or going to weird values. When I checked that I found that the Frsky protocol is a one-way protocol with no checksum. 

Because os that LTM or other ligh protocols with checksum are much better for telemetry. 

As a solution, a checksum could be incapsulated inside Frsky as one of its parameters, but is much better to use opensource code as LTM.

bsongis, 2016-04-27:
Of course yes there is a CRC in FrSky protocol

sppnk, 2016-04-27:
Well, I have er9x firmware in my 9x Turnigy, and the data from cleanflight controllers, included iNav, simulating Frsky, or Multiwii, is not consistent. There is no CRC, as you can see in the document above.

bsongis, 2016-04-27:
The latest telemetry protocol (S.PORT) has a CRC

sppnk, 2016-04-27:
OK, thks. In Cleanflight or iNav you can select Frsky or S.Port. I am talking about the first one. That one has no CRC. 

Are there any firmware for Turnigy 9x modded with S.Port display?

marksev1, 2016-04-28:
Yeah but still s.port is 9600 baud right while ltm packs all the data into a 1200baud, so there are range advances to be had. Ltm is like quite popular protocol for autopilots nowadays. More so that MSP or mavlink.

schwabe, 2016-04-28:
@marksev1 baud rate and protocol format are unrelated. S.Port on the taranis and the s.port of the receiver are 125k baud iirc. On the air the protocol is trasmitted as part of the rf signal and uses the baudrate of whatever that signal uses.

Also as @bsongis stated, OpenTX is not against implementing the protocol, just someone has to implement that protocol and open a pull request.

marksev1, 2016-04-28:
Yes I meant the over the air baudrate. Range improvements could be had with ltm.

projectkk2glider, 2016-04-28:
@marksev1 Are you aware that OpenTX has no influence on the over-the-air data communication protocol. That is under FrSky proprietary firmware control. How do you plant to get the protocol data from the model to the radio?

marksev1, 2016-04-28:
I know that opentx doesnt have influence. but supporting ltm for telemetry link would increase range of rc link...I would have it connected via a serial port connected between model and radio? Module does of course not need to be Frsky, but can be an Openlrsng module.

sppnk, 2016-04-28:
@marksev1 :+1: 

I am searching for a method to have LTM telemetry on ground, and having it in the screen of my 9x transmitter would be amazing. I dont't know exactly how much can be lowered the baud rate in openlrsng system, but I guess it can be configured in chrome. This LTM thing would be a great advance for fpv long range flyiers

sppnk, 2017-02-08:
Old thread, but for info: I have had to use a bluetooth link from TX to have LTM  in my phone. For now is teh best solution. I dont know if there is a transmitter reading LTM. For iNAV users this should be a good thing.

closed 2019-11-15 by stale[bot]