opentx/opentx #3387: Intermittent issue blank screen, no audio

puptv, 2016-03-27:
After upgrading to the opentx-taranisplus-lua-en 2.1.7 version I started having intermittent issues with audio playback failing, the screen going blank, and in one case, the aircraft becoming uncontrollable. In all cases I have found that cycling the power corrects the issue.

I have reloaded the firmware with the non LUA version, reset the transmitter to see if I can isolate the cause.

My question at this time is if there is some sort of debug mode or logging I should enable to help track down the trouble. Any other quetions, ideas or suggestions?

Thanks

bsongis, 2016-03-27:
Do you use both internal and external module? Which protocol(s)?

kilrah, 2016-03-27:
Which method did you use to flash 2.1.7?

puptv, 2016-03-27:
I'm just using the internal module communicating to a FRSky XSR-SB with 14 channels of data. I believe I also saw the problem when using a X8R.

I upgrade from FRSky's default firmware to 2.1.7 using the Companion 2.1 software. Additionally, I added the sound files.

kilrah, 2016-03-28:
Yeah but what method did you use to flash? Through USB In bootloader mode, through USB powered off, by copying the file on the SD card?
Asking this since we've had a couple of reports about the first of those sometimes leading to a bad flash. If the firmware wasn't written properly and bits are missing it can crash unexpectedly.
Please state what version you had on the radio before upgrading too.

bsongis, 2016-03-28:
You don't have any DEBUG function enabled?

puptv, 2016-03-28:
I am pretty sure I originally uploaded the software with the Taranis-Plus powered off. Recently I've loaded the non-lua version in bootloader mode via the USB, and now through Companion software with the transmitter powered off.

I'm not exactly sure what FRSky production firmware was on the Taranis when I received it. I assume it was the most recent version since I just purchased the radio a few months ago, and it seemed like that software hadn't been updated in a couple years when I checked.

I am not running the x9d in debug mode. How do I enable the debug mode? I don't see it as a build option.

Thanks

Daniel

bsongis, 2016-03-28:
Would you paste your EEPROM content here? And would you tell us on which model this happens? All?

bsongis, 2016-03-28:
When it happens, is the RX on?

puptv, 2016-03-28:
Sure...  I was working to double check that the problem still happens as I remove functionality so I could share a much more indicative problem report.

Yes, it is happening with all of my MultiRotors which all run a SP Racing F3 controllers and the XSR-SB receiver with telemetry.

The challenge is that the problem has been very intermittent...  Possibly being related to hitting the SH down switch a lot to hear the VFAS voltage.

Thanks

puptv, 2016-03-28:
I would be happy to attach the file, but all I get is: "Unfortunately, we don’t support that file type." even with a zip file.

kilrah, 2016-03-29:
Just use wetransfer or any similar service to upload and post the link. 

Is the screen always blank after the event? If not can you see the [!] icon?

bsongis, 2016-03-29:
It would be interesting to know if there are serial errors over the S.PORT bus. I will check if we have a counter there

puptv, 2016-03-29:
Update: I reloaded the 2.1.7 non-lua software repeatedly, including once today with a power off usb transfer and the evidence of the intermittent issue has currently disappeared. Unfortunately I have been changing many other things and the issue has been so random that I am not yet convinced it's fixed.

I'm currently looking into restoring an old bin file.

--

Is it possible to modify the companion software to detect errors or perhaps even disable power-on programming? 

Thanks

Daniel

puptv, 2016-03-29:
daniel@puptv.com
sent you some files

‘WeTransfer File.’

Download:

https://www.wetransfer.com/downloads/05ca1f5dcc87a2249af69fc18f84776120160329054924/8000f96ceca9026033bf6f50b774dbc220160329054924/2fd77f

Files (7.17 KB total)
6PostSmartLowBattery.zip

Will be deleted on
5 April, 2016

Get more out of WeTransfer, get Plus
https://www.wetransfer.com/plus?trk=email-recipient_download

About WeTransfer: https://www.wetransfer.com/about
Contact: https://www.wetransfer.com/contact
Legal: https://www.wetransfer.com/legal

To make sure you can receive our emails, please add noreply@wetransfer.com to your [trusted contacts](http://we.tl/trustedcontacts)

projectkk2glider, 2016-03-29:
Kilrah was talking about this icon (left of the speaker volume indicator):
[image]

Did you notice it?

puptv, 2016-03-29:
I do not see the [!] icon in either the sound or the screen going blank. Anyone's guess if I would have seen it when the screen went blank. In the one case where there was a catastrophic loss of communication I do not recall seeing it.

projectkk2glider, 2016-03-29:
This could also be a hardware problem that just coincidentally happened around the same time that you upgraded your radio. We can not rule out this option jet. To prove it you would have to go back to the previous version of firmware and really check that the same problem does not happen.

puptv, 2016-03-29:
daniel@puptv.com
sent you some files

‘I'm inclined to see if I can bracket the issue...

I just got it to fail again using this original bin file. 

I'm going to see if I can slowly change features and find what makes the critical difference.’

Download:

https://www.wetransfer.com/downloads/9a7004bdae19fb31ba4668b98d9ca9c920160329205556/164357c1da388467b6419f166f7f2cbf20160329205556/b6bc13

Files (2.53 KB total)
Pre-ResetForDebug.zip

Will be deleted on
5 April, 2016

Get more out of WeTransfer, get Plus
https://www.wetransfer.com/plus?trk=email-recipient_download

About WeTransfer: https://www.wetransfer.com/about
Contact: https://www.wetransfer.com/contact
Legal: https://www.wetransfer.com/legal

To make sure you can receive our emails, please add noreply@wetransfer.com to your [trusted contacts](http://we.tl/trustedcontacts)

puptv, 2016-03-29:
During testing today I discovered the radio indeed did have a hardware issue. The plug going to the speaker board had gotten loose.

Thank you for your help. It indeed seemed like a software issue.

bsongis, 2016-03-30:
And does it fix the problem completely if you plug it correctly?

puptv, 2016-03-30:
Yes.  I didn't want to close the issue until I tested it. Because of the intermittent nature of the problem, it could still exist, but my tests with a configuration that previously caused problems worked perfectly this afternoon.  

Again, I thank you all for the great work you've done. I had hoped to give back to the community by helping to track down an intermittent bug, but am also glad to find out that it was a hardware issue.

closed 2016-03-30 by puptv