opentx/opentx #3392: Distorted slow audio on 1 model

fraternl, 2016-03-30:
When I started using a model again it suddenly has distorted slow audio as if the device is too busy doing something else

[image]
[image]

fraternl, 2016-03-30:
I had no time to investigate this. Will do later, but wanted to save everything that might be needed to reproduce this.

projectkk2glider, 2016-03-30:
It's an old problem that shows it self once in a while. Some related issues are #2020,  #1779, #894 and #894.

lshems, 2016-03-30:
Same here. I've come to live with it, as I just run very extensive lua scripts, and they ALWAYS have my radio stutter.

I've recently found the debug statistics screen on the transmitter, by getting into statistics and pressing up.

Could anybody explain what these LUA parameters are, and what normal values should be??

Thanks.

bsongis, 2016-03-30:
@lshems Do you mean you have one script which always reproduces this sound problem? This is VERY interesting is this is the case. Would it be possible you share this script? Lua scripts are executed in the main task which has a lower priority than the audio. That's said if your script uses a lot the SD card, it will take the mutex on DISKIO and will block the AUDIO task!

lshems, 2016-03-31:
Any explanation on the debug Screen?

Ill see if i can make a Nice example to reproduce.

projectkk2glider, 2016-03-31:
**Lua duration**: the maximum time it took to execute one *loop* of all Lua scripts.
**Lua interval**: the maximum time between calls to the Lua *loop*.  Bacically a measure how ofter the scrip `run()` function is called.

Both of these can be reset any time by pressing ENT on that screen. The numbers will rise with the script complexity.

lshems, 2016-03-31:
Ok,

I got timings of 900-1300 for both duration and interval at the critical parts where stuttering is audible.
When stuttering has passed, due to finish of loading the script, it lowers to 30-140 for both.

I am almost sure it is not only file loading, but also calculation time.

lshems, 2016-03-31:
I thought that every call to run() or background() needed to be finished in 20 ms??

So I do not understand the results and explanation. 

What is a loop ?? I have only 1 lua telemetry script running at the same time ?? It cannot take 900-1300 ms to finish, can it?

projectkk2glider, 2016-03-31:
A loop is one iteration of calls to all active Lua scripts `run()` or `background()` functions.

> It cannot take 900-1300 ms to finish, can it?

Apparently it does take such a long time (remember this is max, so at least one loop took that long). The script execution time can be prolonged if Lua is waiting for free SD card mutex. That can for example happen when other threads (Audio or Telemetry logs) are using the card. The time spent in OpenTX Lua functions does not count against the script allowed instruction count. Only the true Lua instructions are counted. All opentx API functions are in fact C functions and they are not counted in Lua.

closed 2016-11-27 by fraternl