opentx/opentx #3393: Horus misc issues

kilrah, 2016-03-31:
- [ ] SWR warning not working

- [ ] Key beeps/haptic missing in "All" mode (could be all platforms, maybe bigger rework due)

- [ ] T5/T6/X/Y inputs missing

- [ ] Quicker method to edit large values with encoder (I think we have something on X9E)

- [ ] "Display Trims" missing (or remove option / make that a layout option)

- [ ] External module PXX not working

- [ ] Linear icons and name fix for L1/L2 (currently S3/S4)

- [ ] "||" missing in "BGMusic ||"

- [ ] Play Track Global Function not saving selected track

- [ ] Play Sound GF not doing anything

projectkk2glider, 2016-03-31:
>Quicker method to edit large values with encoder (I think we have something on X9E)

An idea: slow scrolling - small steps like now. Fast scrolling - big steps.

kilrah, 2016-04-01:
How do we handle T5 and T6? Separate single-use registers, or per flight mode as the others?

projectkk2glider, 2016-04-01:
I would prefer per flight. This gives us in effect another two gvars that can easily be adjusted with the new trim switches without requiring special functions AjustGvar.

kilrah, 2016-04-01:
OK, I started something that way, but then thought it would be better to think more about it...

Some parts of the trim handling are tied to the fact there's only one per control, those aux ones would get a bit out of place, will see what I can think of for some things...
For example it would make sense to be able to select T5/T6 in inputs, but it would not make much sense having them on the flight mode screen as this is stick-related

projectkk2glider, 2016-04-01:
Right, then another simpler possibility is to just have them as controls and do not assign them any backing register. Then user will have to create a special function to use it.

kilrah, 2016-04-02:
Many updates

bsongis, 2016-04-02:
About "Splash time / disable" ... We use this time to load things in the background ...

kilrah, 2016-04-02:
OK I wasn't sure if there was an additional delay, if the time it takes now is all due to loading then just leave it like that. 
I haven't timed, but strangely it feels like that time is a bit variable from one boot to another...

bsongis, 2016-04-02:
I think there is a timeout that could be reduced, and we could add a sort of slider

kilrah, 2016-04-02:
A few updates to the list, and I think I've got the trims quite well sorted (went for full implementation in the end, it actually makes quite some sense) [here](https://github.com/opentx/opentx/commit/1f6a711d24d895d5ac4ba5039574a1acaaba336e).

Not having much luck with the simu interface for it though...

kilrah, 2016-04-09:
Some updates

bsongis, 2016-06-11:
Some updates

kilrah, 2016-06-11:
Updates / reordering

kilrah, 2016-06-23:
Updated

kilrah, 2016-08-30:
Updates made

1Sven, 2017-04-30:
Very impotant things! For me, Accelerometer support may be the first. But one of the Horus futures are not listet: The Musikplayer. Of corse many people means thats not imprtent. But I fly oftens with music and for training it is usefull per switch any music activate. Impotent is that the music never kills the normal sounds of the Horus. I hope it will coms with the 2.21?

kilrah, 2017-04-30:
Music playback has been supported in OpenTX for about 4 years.

1Sven, 2017-04-30:
Realy. But it is very bad to use. If the hardwareplayer from Horus present again, you can play mp3 in stereo.
This is a great difference to the mono-special-designet *.wav- files...
It is posibele I think. But the work about it is extreme...

kilrah, 2017-04-30:
The MP3 decoder is used to play the sound mix in OpenTX so is not available for actually playing MP3s.

The entire sound system would have to be redone, so that won't happen anytime soon.

1Sven, 2017-04-30:
l now. That is the Problem... Take the sound of mp3 in frsky horus and you will hear the difference to the sound in otx.  Absolut happy with otx,  but it will be give things that goes in better. Whay not? Only to many work is the answer. Oft corse, otx is the very best, eaven without mp3..

bsongis, 2017-04-30:
Except stereo feature, the sound quality should be exactly the same. OpenTX does support s16le (16bits sampling) wav at 32kHz. If you don't know how to encode such files please give here a link to your favorite MP3 and I will prepare it for you.

Support the MP3 codec is only interesting if you want to save SD bandwidth, which is not needed here.

1Sven, 2017-05-01:
Oh, thank you ! It is so grat that every body on openTX is helping, I think a link to a way to make my own long music-sounds is better as at every time us your time. But for an excample it will be nice. So the file atached...

And ... Very thank you!
[ACDC - Back In Black (No Bull).zip](https://github.com/opentx/opentx/files/968361/ACDC.-.Back.In.Black.No.Bull.zip)

mpaperno, 2017-05-01:
🤘

WimSanders, 2017-05-04:
I’m new on this forum, I both a Horus a week ago.
Flashed my Horus 2 days ago with openTX.
Main reason: use of existing D8X receivers.
This dined work in Horus software, works now fine with OpenTX.
Thanks for that.
Now I find a miner problem, but it confused me.
During charging the green LED stop blinking.
So the battery was loaded! But no after 15 minutes the battery was empty.
Now I find that the LED stop blinking at 9,5V.
Just keep on charging, solve the problem.

With kind regards
Wim

1Sven, 2017-05-04:
On Horus the LED is blinking only for a short time. Load the TX over night and the LED will be dark,
Than is the TX rady for use.

WimSanders, 2017-05-04:
Sorry, stupid me!
RFM!!

bsongis, 2017-05-06:
Input #0, mp3, from './ACDC - Back In Black (No Bull).mp3':
  Metadata:
    artist          : ACDC
    album           : No Bull
    title           : Back In Black (No Bull)
    genre           : Rock
    lyrics-ger      : Back in black
                    : I hit the sack
                    : I've been too long I'm glad to be back
                    : Yes I'm let loose
                    : From the noose
                    : That's kept me hanging about
                    : Lookin' at the sky
                    : 'Cause it's gettin' me high
                    : Forget the hearse 'cause I'll never die
                    : I got nine lives
                    : Cat's eyes
                    : Abusin' every one of them and running wild
                    : 
                    : 'Cause I'm back
                    : Yes I'm back
                    : Well I'm back
                    : Yes I'm back
                    : Well I'm hey hey hey hey
                    : (Hey hey hey hey)
                    : Well I'm back in black
                    : Yes I'm back in black
                    : 
                    : Back in the back
                    : Of a Cadillac
                    : Number one with a bullet
                    : I'm a power pack
                    : Yes, I'm in a band, with a gang
                    : They've got to catch me if they want me to hang
                    : Cause I'm back on the track
                    : And I'm beatin' the flack
                    : Nobody's gonna get me on another rap
                    : So look at me now
                    : I'm just makin' my play
                    : Don't try to push your luck
                    : Just get out of my way
                    : 
                    : 'Cause I'm back
                    : Yes I'm back
                    : Well I'm back
                    : Yes I'm back
                    : Well I'm hey hey hey hey
                    : (Hey hey hey hey)
                    : Well I'm back in black
                    : Yes I'm back in black
                    : 
                    : Well I'm back, Yes I'm back
                    : Well I'm back, Yes I'm back
                    : Well I'm hey hey hey hey
                    : (Hey hey hey hey)
                    : Well I'm back in black
                    : Yes I'm back in black
                    : 
                    : Oh yeah
                    : Oh yeah
                    : Yes I am
                    : Hey yeah, yeah oh yeah
                    : Back in now
                    : Well I'm hey hey hey hey, I'm back
                    : Hey hey hey hey, (I'm back)
                    : Hey hey hey hey, (I'm back)
                    : Hey hey hey hey, (I'm back)
                    : Hey hey hey hey, (I'm back)
                    : Hey hey hey hey
                    : Back in black
                    : Yes I'm back in black
                    : 
                    : Out of sight
    track           : 1
    comment         : www.radio-fx.de
    publisher       : ROCK ANTENNE
    TRSN            : ROCK ANTENNE
    encoded_by      : Tobit.Software, Germany
    TBPM            : 93
    date            : 2008-04-02 16:53
  Duration: 00:04:04.84, start: 0.000000, bitrate: 128 kb/s
    Stream #0:0: Audio: mp3, 44100 Hz, stereo, s16p, 128 kb/s

1Sven, 2017-05-06:
Oh, your very good!! I like this sound, but never I understund the text. (Now it's going on...) Today I was flight with a 4m glider, and the figures I made exakt on this sound. It makes me happy... I have seen that any things may be possibele on tx. (MP3?) But It is the absolute right way not to implement new futures bevore the targets for 2.2 are completed! Thank you, more I can't say and do....

bsongis, 2017-05-07:
The data was inside your file, it would even be possible to display them on the Tx, but I am afraid it will be useful only once you have broken all your planes.

By the way here is the .wav conversion:
[s16le_32kHz.zip](https://github.com/opentx/opentx/files/981400/s16le_32kHz.zip)

1Sven, 2017-05-07:
Oh, thank you! I hope, that newer all my planes will distroyed... For now and in the future, I can convert the *wav self.  In a hopevully future I think, MP3 is for Horus usable again. There was people with outstanding ideas for that to reanimate the killed MP3- hardware, I've seen in in the chat.
Thank you for the help!

bsongis, 2017-06-22:
You didn't answer, is the quality good? Playing MP3 means we cannot mix any prompt or beep over the background music, I see no benefit in adding this feature

WimSanders, 2017-06-22:
Dear Bernard,

In the beginning when I both my Horus, and with the FrSky software MP3 paying was possible,

I was very surprised, and think this is a nice option.

But now when I’m staring to use the Horus I think there is no need for this in the field.

It only deduce from flying.

So you are right, it has total no added value. Even when you can’t play  warnings during MP3 is a total disadvantage.

with kind regards,

Wim.

lthole, 2017-08-09:
Has the screen for GVARS redesign #3185 (additional paramaters) been implemented? I see it in X9D+

closed 2020-05-01 by kilrah