opentx/opentx #3394: Companion 2.1.7 won't open 2.0 Taranis EEPROMs

kilrah, 2016-04-01:
Nor any that have been converted from 2.0 to 2.1 by the radio.

Seen the issue reported a couple of times including [here](http://openrcforums.com/forum/viewtopic.php?f=28&t=8668&p=113582#p113582) and confirmed.

[image]

kilrah, 2016-04-01:
2.1.6 and 2.1.5 have the same problem

techcoast, 2016-04-05:
I just bought my Taranis, with a 2.1.6 build and even with the same companion I have the same issue. I can send/receive straight to the SD card, but I can't get companion to load from the tx without that issue.

techcoast, 2016-04-05:
Well, apparently upgrading to 2.1.7 on companion fixed the issue.

stormhigh, 2016-06-12:
Have the same problem - with Companion 2.1.8. writing in Taranis X9D as it should. Reading models from is impossible - Failure-message same as above. Any solution till now?

Greets and nice time!

stormhigh, 2016-06-12:
Not solved - but better... just restored eeprom by taranis on SD-Card. Open it from SD-Card with Companion and I have everything to save, modify, whatever, and can write it back to Taranis with usual button in Companion.
Hope this helps. Take care!

rhuitl, 2016-10-09:
I would suggest to rename this ticket to "Companion 2.1.5+ won't open Taranis EEPROMs" because it happens for me with 2.1.9 after formatting the EEPROM (with OpenTX 2.1.9) and @kilrah reported it to occur on 2.1.5 and 2.1.6. I cannot open the EEPROM file from the "TARANIS" USB mass storage device, either.

There is a workaround that I think @stormhigh was using although it was a bit unclear to me how to do it. On the Taranis, long-press MENU, go to the VERSION screen and select backup the EEPROM to SD-Card. This file is 32768 byte in size and can be opened by Companion.

The EEPROM file on the "TARANIS" USB mass storage device, however, is 65536 byte in size, just like the file downloaded by "Read Models and Settings From Radio".

kilrah, 2016-10-09:
What bootloader do you have? The file on the TARANIS drive should absolutely not be 65536 bytes, but 32768. First time we hear of the issue. You might have the broken FrSky 2.1.6 bootloader some radios were delivered with.

rhuitl, 2016-10-09:
I had opentx-x9d-v1.0.01 on this Taranis which I bought in August 2013. Today I updated it to 2.1.6 from the FRSky homepage which got me bootloader 2.1.6. Because this FW was freezing in the model selection menu, I updated to 2.1.9 (bootloader is still at 2.1.6).

tomkfly, 2016-10-09:
Try updating the Bootloader.
Read http://www.rcgroups.com/forums/showpost.php?p=35887149&postcount=49858

rhuitl, 2016-10-09:
Awesome, updated BL to 2.1.9 from the SD-Card page in OpenTX (great feature!) and that fixed the problem with "Read Models and Settings From Radio". It is fully working now (and the file is 32768 bytes in size). Thanks a lot!

closed 2020-05-01 by kilrah