opentx/opentx #3397: Include LUA Table library

rcbebo82, 2016-04-01:
It would be nice to have the table library available. It would be useful for working with Lipo cell sensor data and gps sensor data.

Currently the only way to get the size of a table is to loop through the table. The lua table library has a function in it all ready: table.getn(Table1)

B.R.
Benjamin Boy

lshems, 2016-04-01:
I don't think you are correct. The table library is really not neccesary.

the size of a table is given by  the "#" operator.
pack and unpack can be done by simple pure LUA functions

"#" has some limitations only when using sparse tables containing "nil"'s, like t={} t[1] t[212]
if using strings as indexes, # is not useful, as getn.

please explain if I'm missing something??

rcbebo82, 2016-04-04:
Could you give an coding example? I think we dont have any "nil"s in tables right at the momement. If this works table library is not needed!

Sacre100, 2016-04-05:
#Table1 is the same as table.getn(Table1)

lshems, 2016-04-05:
if it is of any help, a sort of mini library for table

```
local table={}

table.pack=function(...)  
  return {n = select('#', ...), ...} 
end  

table.unpack=function(t, i , j)  
  i = i or 1
  j = j or t.n or #t
  if t[i] and i<=j then  
    return t[i], table.unpack(t, i + 1 , j)  
  end  
end  

-- Print contents of `tbl`, with indentation.
-- indent sets the initial level of indentation.
--set indent to "tableName: " to be able to see which table is printed if you use it at several places in your code

table.print (tbl, indent)
  if not tbl then print( "empty") return end
  local formatting
  
  if not indent then indent = "   " end
  for k, v in pairs(tbl) do
    formatting = indent .. k .. ": "
    if type(v) == "table" then
      print(formatting)
      table.print(v, indent .. "   ")
    else
      print(formatting .. tostring(v))
    end
  end
end   

```

rcbebo82, 2016-04-05:
Table Library not neccesary: 
cellResult1 = getValue("Cels")
lcd.drawText(lcd.getLastPos() + 1, 55, ' | ' .. #cellResult1 .. 'S', 0)

Works fine!

closed 2016-04-06 by projectkk2glider