opentx/opentx #3398: Wizard "throttle cut" shouldn't use Override SF

kilrah, 2016-04-01:
Otherwise people with the "nooverridech" option can't see nor remove it if they selected it during model creation.

fraternl, 2016-04-11:
What am I missing...???
Why is there an option "nooverridech" ???

jtaylor2, 2016-04-11:
>@fraternl
>What am I missing...???
>Why is there an option "nooverridech" ???

Because it is not needed and if used incorrectly can damage linkage and servos.
See issues 1428 and 609.
https://github.com/opentx/opentx/issues/1428
 https://github.com/opentx/opentx/issues/609

Jim

fraternl, 2016-04-12:
I read those tickets and it didn't convince me...
Someone already predicted a compatibility issue and here we have one now...
I don't have and will probably never have a helicopter. I checked the "noheli" option because it's there...
If they would take that option out I wouldn't care....

Half of these compiler options could be scratched imho.
Only if they affect memory usage they should be in there....
FAIMODE and FAICHOICE are good exceptions.

It also complicates programming and could be a source for bugs like this one...

lshems, 2016-04-12:
I think compiler options are nice for maintaining backwards compatibility. In case of the channel override, the only real issue is that override channel seems to force a ppm value, passing everything else. Then it should probably be defined as ppm value as well in the setup. That would make it very clear in its usage.

Perhaps take that into account with the general discussion about units: gvar (raw), %, ppm,....

fraternl, 2016-04-12:
These options are not there for compatibility... 
The nogvar, for instance, is created by request of users that don't want that extra page.
imho this could better be done at runtime....

It's only because they have this compile on demand that it's possible to do this way.
IMHO it's not appropriate to use it for this....
Only when it saves resources for the system...

timer3, noheli, nogvars, mixersmon, nooverridech   should (IMHO) not be a complier option. 
It leads to misunderstandings when someone is using foreign setups or advice and can also lead to bugs.
I am sure the developers are aware of this.
I wonder if the people asking for it are.

kilrah, 2016-04-12:
> These options are not there for compatibility... 

In a way yes, because when the nooverridech option was discussed it was otherwise going to be removed altogether and would have broken nearly ever model out there. I personally never understood that one, even less than the others. 

Agreed options would be better implemented as general settings, we've actually said that "we should do it one day" since we started working on the Taranis 3+ years ago. But reusing the existing options system was more adequate given the time we had at hand back then than reworking the menus system, and now given it works and there's more important stuff to do it's stayed that way. Maybe you could do the job?

fraternl, 2016-04-12:
I'm not aware of the full history and motives behind it.
I'm just observing as it is now.
I fully understand it if choices are made for other reasons than the best way.

> Maybe you could do the job?

Although I was a programmer and am still doing a lot of creative stuff with bash (yeah right)... I'm not adequate enough in this field. I used to program stuff that involved databases and administrative processes. I stopped doing RC in that time (30 years ago).
I find your work much more interesting than what I did, but that's just choices in life.

lshems, 2016-04-12:
Since it's not only about adding a line of code anymore, but also being able to set up the entire toolchain to have the code compiled, opensource is not really a good descriptor of the way a lot of opensource projects are run. 

This is no criticism in the direct sense, that I blame someone or some group. It's just that it has become impossible for us mortals to add just a line of code and test it ourselves.

I think the same goes for Frater, but i cannot speak for him.

schwabe, 2016-04-12:
@lshems While that may be case, OpenTX was one of the more easy projects to setup. Mac OS X setup was a few commands to install compilers, etc. and then everything worked. Sure I have a computer science/programming background but without that, coding on complex projects like OpenTX is difficult anyway. Especially for OpenTX I think blaming the setup/toolchain is unfair.

lshems, 2016-04-12:
@schwabe Sorry, but I don't like it when the impression is given that I blamed the setup/toolchain, and  that would be unfair, in other words: that I would be unfair. 
I tried to be as gentle as can be in making clear that I don't blame anyone or anything, but that there is a "thing" which prevents "us mortals" to add something to the code base.

Especially this subject is nice, since there is only ONE change to be made to only ONE digit in the entire code to get what I ask for; It is changing the limit for countdown from the hardcoded 10 seconds (100 units of 1/10ths of a seconds to be precise) to 20 seconds.

If I could get the toolchain up and running, yes, I could compile myself and test this code change, and then even get motivated to try and get some things fiddled in to have it a real parameter.

So yes, the toolchain plays a big role here, especcially when forced to work on windows, but it isn't the only thing.
 
As said previously, i'm not blaming anyone, just making an observation which I believe is true for "us mortals".

It's a pity nonetheless.

fraternl, 2016-04-12:
> So yes, the toolchain plays a big role here, especcially when forced to work on windows,

Without 'vi' I can't do it....   ;-)

bsongis, 2016-04-12:
Guys we are setting up Docker containers, so that there will be soon no excuse if you cannot compile your firmware yourself :) I think the only remaining task is to write the wiki, but really it will be a child game ...

schwabe, 2016-04-12:
@lshems I just wanted to share my experience. I started with OpenTX coding just 2 months ago.

bsongis, 2016-04-12:
@schwabe you are a developer, that helps for sure :)

fraternl, 2016-04-12:
but on topic....
Is it an option to remove the "nooverridechannel"???

lshems, 2016-04-13:
I'm already happy with the other info. If that comes available, I'll probably dive into the opentxcode myself,

lshems, 2016-04-13:
@schwabe no offence taken. Just pay attention on how to formulate things when 'quoting' others. I would certainly appreciate it not being put words in my mouth I didn't use.

:)

schwabe, 2016-04-13:
@lshems sorry, for misinterpreting your comment.

lshems, 2016-04-13:
Appreciated! I really wish I could get into developing, but I spent several days trying to get it to work, and I simply don't succeed. 

That is probably mainly due to my lack of knowledge, so I'm very happy to here there are things in progress.

kilrah, 2016-04-13:
Pretty easy to do in a Linux VM.

lshems, 2016-04-13:
Yeah yeah and yeah. All is easy, if you have done it ONCE. it's the first time that you get stuck on all small details and missing one somewhere.

But I think the discussion on "pretty easy" is a dead one with respect to OpenTX. I'v been told that everything is easy, and yet see a thread on RCGROUPS with more then 50 messages a day, half of which are questions.

Just observing, not judging.

P.S. I know how to set up a virtual machine. Tried both ways. Direct on Windows, and in a VM. Oh yeah, and I did tell already I spent several days on it, and it is my fault I don't get it to work. I'm just not the guy for simple things.

LOL LOL LOL

lshems, 2016-04-13:
@fraternl  Sorry for robbing your issue. DIdn't intend to.

kilrah, 2016-04-13:
For linux there are instructions that take about 10 steps, while for windows there are no up to date instructions and if there were some they would be 50 steps. So yes it's easy in comparison.

Can you maybe cut the crap for once?

AlgebraicMirror, 2018-04-02:
For what it's worth, I agree with removing the "nooverridechannel" compiler switch, and making it a global setting on the radio itself that users could turn on and off on demand. If a user wants to have it disabled for perceived safety then they can, but they can at least turn it off at runtime if they have a model where they need to make that modification, and can't get access to companion.

I do not agree with the suggestion to not implement throttle cut with channel override though. It is a very natural way to do it, and probably the best way to do what Spektrum currently does today - they do a channel override of -130 (in Spektrum terms). If you are flying any of the common Spektrum BNF planes, which many of us are, it is very useful to have this feature available in the new model wizard, and I wouldn't want to see it removed entirely. 

I suppose if you wanted to get really fancy you could hide the option in the wizard if you detect their firmware doesn't support it, but I think the best and most straightforward solution is to just get rid of the compile option and move it to a global setting, so that all radios can support the feature.

closed 2020-05-01 by kilrah