opentx/opentx #3400: Offsets don't show on Inputs headlines

tomkfly, 2016-04-03:
When an Offset is used in mixes it shows in the mix line, but not in Inputs. This caught me out today as when I flicked the rate switch Maidening a new model model which was very sensitive on Aileron I flicked the rate switch and it started to roll. Fortunately It was 3 mistakes high. An offset had crept into my Aileron Input mid rate (I must have somehow entered it accidently when setting up the Inputs) and on checking through all my settings I missed it, as I didn't realise offsets didn't show. Not a priority, but I think it would be a useful enhancement if offsets could be listed on the Input line.

projectkk2glider, 2016-04-03:
If you mean in Companion, then already fixed #3357. Is it on the radio?

tomkfly, 2016-04-03:
Radio and Companion 2.1.7

projectkk2glider, 2016-04-03:
Just checked. The `offset` on the radio is not shown in both Inputs and Mixes. I am sure this is because of the space limitations. @bsongis @kilrah @mhotar can we do something to squeeze the offset there?

Current display:

[image]
[image]

kilrah, 2016-04-04:
Offsets have never been shown on either screen, indeed because the available space is used for other things.

closed 2020-07-08 by stale[bot]