opentx/opentx #3402: Fuel timer

franz4949, 2016-04-04:
Hello,

I am looking for having a flexible control over the running time of a fuel operated motor. OpenTx offers in the model setup page the timer option Th% to count backwards from a start value down to zero with variable countdown velocity analogue to the throttle stick position. That's fine but I would like to see an option for making it adjustable to a specific fuel consumption. Let's assume that with filled up fuel tank and all the time full throttle operation the motor will have a total running time of 10 minutes, if operated all time in minimum throttle position the motor will have a total running time of  20 minutes. So I would like to have an additional timer option, let's say ThF% and the possibility to set the two time values corresponding to full (10 min) and minimum (20 min) throttle position and a switch option (mechanical or logical) for activation of the countdown. With that input values the timer shall work analogue to the already existing Th% function however counting down on a linear function basis with the values specified. Timer start value is automatically the setvalue for throttle min position, so for aforementioned values here 20 min. The countdown shall start/stop with the specified activation switch and progressively count down with the throttle stick position.

projectkk2glider, 2016-04-04:
This would be a perfect job for Lua script. Ask on forums, for example http://www.rcgroups.com/forums/showthread.php?t=2180477

kilrah, 2016-04-04:
But then you'd be assuming the fuel consumption is linear with throttle, which it usually isn't either!

Anyway you can use a free channel as Throttle source (it will then be the source of the TH% counter), and add a custom curve there with any response you like.

closed 2016-04-04 by kilrah

lshems, 2016-04-04:
PM me on rcgroups, i'll see if I can get it done in a LUA script.

kilrah, 2016-04-04:
As I implied a script isn't even needed.

lshems, 2016-04-04:
Not needed, but perhaps more userfriendly :)

I can swim to England, but I prefer a ferry.

franz4949, 2016-04-04:
I would see a modification of the already existing Th% timer function as a useful and easy to setup improvement (and even if the fuel consumption may not be exactly linear inbetween the throttle stick end positions). Maybe you reconsider to put such modification on the wishlist. Thanks anyway to everybody who commented so far.

projectkk2glider, 2016-04-04:
I am against adding such features:
* that are used by minority of users
* can be implemented easily using existing features (two ways in this particular case)

franz4949, 2016-04-05:
I respect your view, but I don't understand how you can say "feature that is used by minority of users"? There are many users out flying with fuel operated motors and turbines and they all would like to have some better control over the fuel consumption. Unfortunately you will not find flow sensors in the RC market except for Jeti and that fow sensor costs 180,- Euro. So a cheaper solution would be highly welcome and I find my proposal, if implemented, sufficiently accurate for the purpose intended. Yes, such timer can also be achieved by the two alternative option that were pointed out. But for many users both options are more complicated to setup and LUA scripting is not for everybody. My proposal would be very easy to use and setup for everyone to match the specific data of his motor (and mostly he will have some planes with different motor data), without any need of programming or scripting. And ease of setup and use I believe is always an important factor.

kilrah, 2016-04-05:
The fact you're the first to ask about it in that many years shows the low interest for sucha feature.
We've had people ask about whether it was possible to do it before, we mentioned the "curve on a free channel" and the answer was always "oh yeah easy, I should have thought about it". Setting up one mixer line, the associated curve and choosing the throttle source isn't *terribly* complicated.

So yeah it's pretty hard to see much worth in the implementation and maintenance efforts needed to make that an integrated function. 
It is also not generic enough for that, and feels like should we proceed with that while YOU find your proposal appropriate you can be sure that 2 days later someone will come and not be happy with it and wanting it done differently.

franz4949, 2016-04-05:
Ok I accept your view and decision and I may inform you that I have been successful with implementation as per your proposal (additional channel mixer line with associated curve and activation of countdown by logical switch). Thank you all for your quick responses and proposals for solution.

lepicfoto, 2016-04-05:
little info fore you :-)
 I know for sure that N2 Industries, looks at getting a flow sensor for a decent price and with a good resolution, and there is support in the telemetry of such a sensor allready. so it is surely only a matter of time before some hardware will works for an ok price, and then we suddenly how much fuel is left in the tank :-)

kilrah, 2016-04-05:
The fact you're the first to ask about it in that many years kind of shows the low interest for such a feature.
We've had people ask about whether it was possible to do it before, we mentioned the "curve on a free channel" and the answer was always "oh yeah easy, I should have thought about it". Setting up one mixer line, the associated curve and choosing the throttle source isn't *terribly* complicated and much more flexible.

So yeah it's pretty hard to see much worth in the implementation and maintenance efforts needed to make that an integrated function. 
It is also not generic enough for that, and feels like should we proceed with that while YOU find your proposal appropriate you can be sure that 2 days later someone will come and not be happy with it and wanting it done differently.

kilrah, 2016-04-05:
Did you get news from them recently? If I remember well they had pretty much given up because the sensor was too imprecise.

lepicfoto, 2016-04-05:
btw I am not a programmer and do not know lua, I would, but I learn from the bottom up, and as such I am missing a good resource to get started.
so I fly on stopwatch,
full tank + WOT = 18 min, fly 18 min, then back on terrafirma.
The utmost respect to all of you OTX people.

lepicfoto, 2016-04-05:
it is true it was very expensive and imprecise,
but he is looking, and I wonder if he wont finds a solution to the problem without having to cost an arm and a leg,
all the gear he's doing really works well, so he'll find out.

But it does not hurt to let him know 
that we are some people waiting,
I have no idea how many we are,
but many, I am sure,

it's a system that works and lives :-)
best respect and a good work all.

franz4949, 2016-04-05:
Thanks for your info regarding N2 Industries. I was on their site and latest info there was from May 2015 saying that in the US they will not continue with the fuel flow sensor due to liability law, so they are looking for someone in the UK but obviously they haven't found anyone todate. So unfortunately nothing in sight and on the market you only find the Jeti one.