opentx/opentx #3408: Backlight intensity not work in special functions

janekx, 2016-04-09:
In special functions we can assign level of backlight intensity to some switch. 
It works only when is lcd off and it turn it on, but not change intensity from setting. 
I think if we talk about intensity it should change intensity every time if the backlight is switched on by key/stick or by this function.

kilrah, 2016-04-09:
What radio? What version?
You should know better...

janekx, 2016-04-09:
sorry updated

janekx, 2016-04-09:
Same at X9E OpenTX 2.1.7

projectkk2glider, 2016-04-09:
I did not know that such function even existed :stuck_out_tongue:

kilrah, 2016-04-09:
OK, apparently the slider was added with the goal to allow selecting the backlight color, but there's no backend code written for that.
Is anyone interested in implementing or do we just remove the slider?

janekx, 2016-04-09:
I think backlight intensity is more usable then color change.  Have on 3pos switch backlight off,  60% and 100% is good service for various ambient conditions.

kilrah, 2016-04-09:
The idea of the color was that you could have easily distinguishable warnings. I really don't see the point of dynamically changing brightness on a transflective screen though. When the backlight is too dim it's because there's enough light not to need it anymore.

janekx, 2016-04-09:
On lower ambient light you not want 100 backlight becuase can be unpleasant

kilrah, 2016-04-09:
I find 50 is good - and as said don't need any more in daylight since daylight does its job so I can actually turn it off.

closed 2019-10-11 by bsongis