opentx/opentx #3411: GV not evaluated properly for a>b or a<b

filago, 2016-04-10:
It appears that the GV value used to determine a Logical Switch with a>b or a<b is ~1/10 the actual GV value.  This is with 2.1.7; I didn't check other versions.

Example:
SF1: ON Adjust GV1 Source S1
L1: a>b, Thr,  S1
L2: a>b, Thr, GV1
L3: a<b, GV1, Thr

I believe L1-L3 are logically equivalent so they should go ON/OFF together, but they don't.  L1 behaves as expected, going ON when Thr exceeds S1 value.  But L2 and L3 go ON when Thr exceeds ~1/10 the S1 value.

kilrah, 2016-04-11:
See here for an explanation:
https://github.com/opentx/opentx/issues/1361

closed 2016-04-11 by kilrah

filago, 2016-04-11:
OK, thanks.  My apologies for not researching to find that this is intended behavior for now.

#3185 indicates that it may be resolved at some point.