opentx/opentx #3412: Toggle option on timer countdown in seconds from 30 seconds down

lshems, 2016-04-12:
In a lot of competition flying, a target landing or exact touchdown needs to be achieved. The standard countdown in seconds starting only at 10 is lacking information.

It would be very nice if we could set the seconds countdown trigger level, so you can choose if it counts down i seconds from 30 down, 25 down or whatever value you choose.

Thanks for considering.

projectkk2glider, 2016-04-12:
You could do it with a Lua mix script. Start value can be a script parameter or even gvar. Much more flexible.

lshems, 2016-04-12:
YesYes, I know. It would not cost much either in terms of lines of code. It would have to be a mixes script, I assume.

It is just that I don't like mixes scripts, since they seem to devour memory, and I cant assign them from within LUA, so users would have to explicitly know how to set up a mixer script. That is not obvious at all.

It is however such a standard request, which asks for 1 parameter, the trigger level for seconds countdown, to be added to the timer structure. The parameter is already in the code, I checked that.

If I could compile myself, I would simply change it there.

Thanks for considering a userfriendly, non lua solution.

bsongis, 2016-04-12:
The problem is not the number of lines of code for this, it's just the EEPROM. We cannot add a counter in the TimerData structure without shifting everything. We have to do it in 'next'. This is a really simple change that's said, we are so busy with Horus ports these days. It's difficult to focus on everything in the same time...

lshems, 2016-04-13:
No problème. I'll make a script in the mean time.

lshems, 2016-06-20:
for those looking for a solution in LUA, add following in a function script:

`local timer={}     --

timer.addSeconds = 1
    
local function run()
  
  timer.play=(model.getTimer(timer.addSeconds-1)).value
  timer.countDown=((model.getTimer(timer.addSeconds-1)).start>0)
  timer.old=timer.old or timer.play
  if not (timer.play==timer.old) then
    timer.played=false
  end
  
  if not timer.played and timer.countDown then
    if timer.play < 30 and timer.play > 10 and not (timer.play==20) then
      playNumber(timer.play,0)
    end
    timer.played=true
  end

  timer.old=timer.play

end  

return { run=run }`

closed 2019-11-15 by stale[bot]