opentx/opentx #3413: Potential model corruption

kilrah, 2016-04-12:
http://openrcforums.com/forum/viewtopic.php?f=96&t=8714&p=114036#p114036

2.0, but might warrant some investigation. No idea as is.

pointu, 2016-04-13:
Hi Kilrah,

Why do u put this in 2.1.8 and not the next release, it will be a great idea for us, i suppose doing this stop the production of 2.1.8

Thanks

kilrah, 2016-04-13:
2.1.8 IS the next release.

But this really seems like an individual random glitch, doubt we can do much about it apart from the existing planned improvement to include CRC checking of all data to detect it.

pointu, 2016-04-13:
2.1.8 IS the next release.
I know and we wait for it for a long time !!!
That's why i propose for this 2.1.9

kilrah, 2016-04-13:
If it is a problem it is very serious and would need fixing ASAP. 
But unless he can reproduce we will move it away.

pointu, 2016-04-13:
ok i understand

bsongis, 2016-04-13:
We need more data about this one.

pointu, 2016-04-13:
that's why i said it can't be affected to a milestone
thanks

kilrah, 2016-04-13:
But putting it to a milestone got the attention of everybody, which was the goal :)

pointu, 2016-04-13:
for sure @kilrah but the first attention we need is the attention of the one who got the problem isn't it ?

kilrah, 2016-04-13:
I meant among the devs.

mpjf01, 2016-04-13:
I'm the "one who got the problem" and you have my attention as well. Despite spending a lot of time trying in the last couple of days I have not been able to make the issue happen again. I am flying blind a bit as I have no real evidence, just a guess really, of the event that caused the problem in the first place. If there's anything else that anyone can suggest that I should do, please let me know and I will do it.

closed 2019-11-15 by stale[bot]