opentx/opentx #3417: Taranis screen not working with FW 2.1.8

mickwpaul, 2016-04-14:
Just updated to FV 2.1.8 and on completion the screen flashes blue once on power up the goes out, no display at all. Switches all seem to work still and I can read the radio and all models are still there.  In bootstrap the display is on and works fine.  I re loaded 2.1.7 and display fault is still there on boot flashes blue with no text and then go out. :(

kilrah, 2016-04-14:
Either you loaded the wrong firmware (plus / non-plus) or the fault is an unrelated hardware problem...

mickwpaul, 2016-04-14:
not sure I opened companion (2.1.7) and it said new update.  I updated to companion 2.1.8.   I downloaded the new firmware (from within companion) and did the burn to TX.  It initially failed, I unchecked the "check hardware compatibility" and it then did the write OK.  But off course it does not want to work properly now.  I have tried putting 2.1.7 back into the radio it writes but the fault is the same. Display flashes blue once and then goes off.  Speech still works and it announces "welcome to open tx".  Any clues how/if I can restore it??

kilrah, 2016-04-14:
Then you indeed have the wrong firmware. Configure companion for a Taranis, not Taranis Plus, then file->Download to grab the correct file.

closed 2016-04-14 by kilrah

mickwpaul, 2016-04-14:
Legend, I was a dill using the PLUS firmware.  Fixed BIG thanks!!!