opentx/opentx #3421: 3-way switch mod?

thucar, 2016-04-15:
I'm running OpenTX on AR9x in my TH9X radio. I've replaced the three 2-way D/R switches with 3-way switches and they work a charm with Ersky9 firmware.

Is there any way to get them to work with OpenTX? And if not, is there any plans for the future that would allow this?

projectkk2glider, 2016-04-15:
We are very busy with the Horus now, but we will keep this issue open for the future. 

You obviously had to use the additional inputs. Any information on that.

thucar, 2016-04-16:
As I mentioned, I'm using the http://www.ar9x.net/ board. It has a lot of I/O broken out and easily accessible. So for Ersky9x all I had to do was connect the third position tab from the switches to EXT1 EXT2 EXT3 inputs on the board, via 2.2k resistors. 

Then all I had to do was to go define the inputs used for each switch in the Hardware menu of Ersky9x. Like this: http://static.rcgroups.net/forums/attachments/3/1/9/8/5/0/a8524018-198-3sw.PNG

closed 2019-11-15 by stale[bot]