opentx/opentx #3422: Switch Warning on Reset Flight

PieroCo, 2016-04-15:
There is no control on the switches when you reset the flight

projectkk2glider, 2016-04-15:
Pleas explain in more words. I can not understand where the problem lies.

PieroCo, 2016-04-15:
If you switch models tells me if a switch is out of place, but if  I reset the flight is not carried out no checks on the switch position. Taranis X9d rev. B OpentX 2.1.7

projectkk2glider, 2016-04-15:
It is supposed to work that way. `Reset Flight` does not trigger switch position check/warning. Why would you want that?

PieroCo, 2016-04-15:
If I finish a flight in the landing phase and start a new flight resetting the flight I would expect it to be made a check for not starting in the landing phase

projectkk2glider, 2016-04-16:
It never worked like that, but I find your proposal interesting. Perhaps and option in model configuration, that would (when enabled) check switches position after flight reset. 

I will leave this open for further discussion.

jrwieland, 2016-07-19:
> _If I finish a flight in the landing phase and start a new flight resetting the flight I would expect it to be made a check for not starting in the landing phase_

Being an electric flyer this is one of the 1st Logical Switch I set out to create.
Base Scenario -   Finished flight - recorded flight information in my phone app.  Prepared next battery (Strap it onto the tray) ,  reset flight (SH) before approaching the Helicopter and connecting the battery.

One of the following tracks plays based on my switch positions 
[SA, SC SD are all "supposed" to be  down, but if i was flying 3D; the flight selector SC may have been left in stunt mode when I landed and cut the power.]
(1) "prefly" _Transmitter pre-flight checks complete it is safe to connect the battery now_
(2) "wrgswtch" _Incorrect switch positions, please recheck switch positions_

BTW: transmitter is 9XD OpenTX v.2.18
I think the use SF & LS triggers is a viable option for those wanting this function, but I can see the point that not everyone would like it as a permanent option in their radio, (many models, scripts, music etc. to store), the use of a specific switch code would limit the ability to set the transmitter to the users desired setup "your transmitter, your way"

I subscribed to this issue, if you would like a copy of my EEPE and custom recordings for this function let me know
Keep up the great work everyone!

projectkk2glider, 2017-01-28:
Apparently this was implemented as a part of 7908eaa406e4d487f0026ad38907da40470399e8 here https://github.com/opentx/opentx/commit/7908eaa406e4d487f0026ad38907da40470399e8#diff-fa8e7dd59bd1fd15895927104c73e269R900

closed 2017-01-28 by projectkk2glider