opentx/opentx #3424: GPS Coordinates Compressed/Incorrect on Telemetry Screen - LUA

jmxp69, 2016-04-16:
GPS coordinates displayed using lcd.drawChannel(80,55,"GPS",0) results in truncated/compressed coordinates on the screen. It shows up the same with the telemetry simulator and with live data.

[image]

kilrah, 2016-04-16:
The GPS display function is meant to display the appropriate 2-line format for the telemtry screens, it's not surprising it doesn't bode well with lua.

You might need to display lat and lon yourself separately the way you want them.

jmxp69, 2016-04-16:
How would one go about parsing lat/lon from the gps sensor?

projectkk2glider, 2016-04-16:
https://opentx.gitbooks.io/opentx-lua-reference-guide/content/handling_gps_sensor_data.html

jmxp69, 2016-04-16:
PERFECT! Looks good so far. Thanks guys. Is there any reason the value can't be displayed correctly without going through all of that?

projectkk2glider, 2016-04-22:
Looks like that function is meant to be used with `LEFT` attribute. Working example (notice that horizontal position will change):

```lua
local function run(event)
  lcd.clear()
  lcd.drawChannel(80,55,"GPS",LEFT)
  return 0
end

return { run=run }
```
[image]

closed 2016-04-22 by projectkk2glider