opentx/opentx #3431: Micro SD Card Not Recognised - 9XR Pro

pyr8, 2016-04-18:
Using latest OpenTx 2.1.8 firmware on 9XR Pro.

Unit does not recognize the Micro-SD card. After initial firmware flash, the card is recognized, all subsequent power cycles of the unit do not recognize the SD card.

Reformatted SD card with FAT32, no change.

kilrah, 2016-04-18:
Tried another card?

pyr8, 2016-04-18:
I have tried both the supplied card and now a brand new Kingston 16gb card.

Sorry I should have specified...the card is not recognized by the remote when browsing within the remote (it will not play sounds on startup, etc.), it is recognized by my computer when connected via USB to the 9XR Pro.

projectkk2glider, 2017-01-14:
Could #4211 be a fix for this also?

closed 2019-11-15 by stale[bot]