opentx/opentx #3435: Throttle inverse not acting on selected throttle source (companion and radio?)

lshems, 2016-04-19:
2.0.19, perhaps also 2.1 since it is not noticed yet??. 

When selecting LS as throttle source in the general screen, and selecting throttleinverse checkbox, it is the stick input from Throttle which is reversed.

If this is intended, why can you select a throttle source?

Confused or bug.

projectkk2glider, 2016-04-22:
`Throttle source` an `throttle inverse` are two separate settings. 

>why can you select a throttle source?

It is used for:
* throttle trace
* THt and TH% timers

I see no problem, please elaborate or close the issue.

lshems, 2016-04-22:
Thanks
 The screen is confusing to me, but understood

closed 2016-04-22 by lshems

kilrah, 2016-04-23:
Actually the throttle warning also follows the Throttle source setting, it probably shouldn't.