opentx/opentx #3438: Audio missing in Mac version of Com

rmcclurg, 2016-04-22:


rmcclurg, 2016-04-22:
I checked the following versions of Companion 2.0.17, 2.0.19, 2.1.7, and 2.1.8. The only version of Companion that plays audio from Special Functions is 2.0.17. This problem was fixed some time ago by including the correct audio libraries in the Mac compile. It appears that at some point after 2.0.17 the compilation for Macs changed and left out the needed libraries.

projectkk2glider, 2016-04-22:
@bsongis this is most likely the result of your MAC build VM.

bsongis, 2016-04-22:
If I am not mistaken I should recompile Qt with Phonon support on the MAC VM. I think it will wait OpenTX 2.2 Companion (Qt5 based) where Phonon is replaced by QtMultimedia and is included in the original packages.

schwabe, 2016-04-22:
Is all sound missing or only play tracks? (E.g. do the trim sounds work)

rmcclurg, 2016-04-22:
Just checked 2.1,8 and trim beeps and "trim center" play fine. The model name plays as well. Looks like it is just play track.

schwabe, 2016-05-31:
I cannot reproduce this. Can you double check that your sd card path is set under the radio profile and the audio tracks you are trying to play are really there?

projectkk2glider, 2016-05-31:
There are two different audio subsystems used in Companion:
 * for listening to the selected track (blue triangle button) in Special function dialog a **Phonon** library (or **QtMultimedia** in 2.2) is used
 * for Simulator a **SDL Audio** library is used

So depending on which library is available the sound could work in both places, only one or none.

schwabe, 2016-05-31:
@projectkk2glider Ah I see now. Yes that blue triangle thing is broken for me as well. The OpenTX 2.2 Companion just works out of the box.

projectkk2glider, 2017-01-14:
@schwabe is this still a problem? I hope not and this can be closed.

kilrah, 2017-01-14:
Should be fixed by #4212

closed 2017-01-14 by kilrah

schwabe, 2017-01-14:
I think 2.1.x builds still have the problem but I don't want to spend time to fix the cmake package script for 2.1