opentx/opentx #3439: 9x Timers only allow countdown  version 2.0.19

kemakama, 2016-04-22:
I have installed the latest version 2.0.19 to a Turnigy 9x.  I have found that the timers now only allow countdown and stop at zero. This is regardless of the settings.  The previous version I had installed was 2.0.15 where the timers worked correctly, count up and count down.

kilrah, 2016-04-23:
Please detail your settings, particularly the configured start value. Try on a new model too.

kemakama, 2016-04-23:
Hi Andre
Thanks for your very quick email.  
I have two Turnigy 9x transmitters, Until recently both running the same opentx version 2.0.15.  I uploaded the latest version to one transmitter only when I thought I had a problem with it.  I can make direct comparisons between the versions easily.
As you suggested I have created a new model on the transmitter using 2.0.19.  On the SETUP screen,  I then changed the timer 1 Off/On to ON.  No other changes were made, time set to 00:00, no check boxes set.  On returning to the main screen, the timer 1 status is ON but the count was set to 00:00 and did not change.
I then returned to the SETUP  screen and set the timer to 05:00, no other changes. Going back to the first screen, Timer 1 was set to 05:00 and started counting down.  When the count reached zero it stopped and did not go negative.
This is the same on both timers.
For me this is a little inconvenient, but I can work around it by setting a high value on the timer.  
Is it possible for me to reload the older version.  If so, could you let me know what to do.
I hope this answers all your questions, just let me know if you need more detail.
Thanks again
Keith Ryder Keith Ryder

Home 0121 353 3036 (Leave message)
Mobile 07854 714357

projectkk2glider, 2016-04-23:
I can not replicate this issue in Companion 2.0.19 Simulator. In simulator everything works (count up, negative values, etc..) I don't have the radio to test on it.

[image]

kemakama, 2016-04-23:
Projectkk2glider.  Thanks for you input.  I have now looked at the simulator which works ok, but sadly it does not agree with the transmitter.  As they say, the only real test is............

Can I go back to the old version?

projectkk2glider, 2016-04-24:
>Can I go back to the old version?

You can't get old version with Companion. There are only two ways:
* you saved previous versions of firmware on your computer and now you can flash them back with Companion.
* you (or somebody else) builds a custom firmware for you using older OpenTX code base.

But in reality we need to solve this problem, because if it proves to be a genuine bug, then it will affect all other 9X users.

kemakama, 2016-04-24:
Projectkk2glider Thanks for the information. 

It seems that no one else has reported this issue. Could it be a problem with my transmitter?

Is it worth me trying a reflash of the transmitter?  (I dont really believe it would work but it may be worth a try.)
I could also load the new version to my other transmitter if you think it will help to confirm that  there is an issue.  

If any more information is needed please let me know.

I will stay with the latest version but just a point of interest.  As you said, Companion saves the firmware to the pc prior to uploading the new program. Where are the files saved and what format are they in (name, extension) so that I can find them.  

Thanks

kilrah, 2016-04-24:
Companion *can* backup the **settings** before flashing (needs to be configured to do so), it does not backup the **firmware** (that you would have as a file anyway because you downloaded it, and it would be where you downloaded it).

kemakama, 2016-07-12:
Hi  Further to this issue.  As I explained I have two Turnigy 9x transmitters. Transmitter A with 2.0.19 which does not work correctly and transmitter B with 2.0.15 which is ok.  Can I save 2.0.15  to my pc from transmitter B and load it transmitter A?  Thanks

kilrah, 2016-07-28:
Yes if they have the same hardware options.

ludekcakl, 2016-08-03:
Hello, I just would like to confirm same issue with Turnigy 9x + latest firmware 2.19. Stopwatch doesn't work in "count up" mode.

kemakama, 2016-08-04:
Ludekcaki  Thank you so much for your comment which is confirmation of the issue I raised on April 22nd, four months ago  It has not yet been resolved and may never be resolved.  My belief is that this version was implemented without it ever being tested on a Turnigy 9x.  This has, to me, been confirmed by the lack of feedback from the developers.  Testing on the simulator is irrelevant when developing software.  The only real test is on the actual hardware.   As I said earlier its an inconvenience and you can work around it, but if I am right, to implement software without proper testing is not acceptable, even in a freeware environment.  I worked in IT programming for a many years and I know that proper testing is essential.  Also, I know that any coder worth his salt would not want to have dodgy software against his name, and his first priority would be to sort out bugs!  I believe that this version is not safe to use.   The timer section of the code must have been 'accidentally modified' by changes elsewhere.  Is there anything else that has been 'accidentally' affected??  I think that until this version has been properly tested and debugged it should be withdrawn.  I have now stopped using it and am using my other transmitter with correctly working timers (version 2.0.15).  I am also converting to Taranis, which so far seems to work ok, and I will stay with the pre-loaded software.

lorenzo61, 2016-08-11:
Hello, I also would like to conform this issue, and related to kemakama's last comment, I tested the firmware 2.0.19 on a Turnigy 9x without any other problem. I loaded a quite complex glider program witch use advanced features like wing camber, snap-flap and butterfly airbrakes, on two separate model and after some month of real (slope soaring) use I am completely satisfied from the soft. The only con is the timer that don't works.....

bsongis, 2016-08-11:
I will have a look

omersiar, 2016-11-08:
I also wanted you to know Timer problem continues with 2.0.20, too. (FlySky 9X)

My timers configuration is like below: 

For Stick THt and for the THR

Both does not start count whether throtle rised or THR switch is on. It was working on previous release that i did not pay attention to version number

kemakama, 2016-11-08:
Hi omersiar.    Many thanks for your comments.  I did consider an upgrade to 2.0.20, but now I wont bother!

omersiar, 2017-01-02:
Latest firmware (and companion) "Version 2.1.9 / 2016-09-14" solves this exact problem. Timers are working now, both count-up and down.

closed 2019-08-19 by kilrah