opentx/opentx #3443: FAI Mode problems in OpenTX 2.1

projectkk2glider, 2016-04-25:
Due to the telemetry changes in 2.1 I believe that the FAI mode is not implemented correctly. Here are the issues I see:

While the radio is in FAI mode:
* receiver voltage and RSSI can not be viewed/played/used in logical switches - it is allowed by the  FAI rules.
* Altitude can be displayed on the main screen (on the top bar)
* All telemetry values are available to Lua scripts
* Lua script can use IO library to read telemetry log and extract values

Since the FAI rules allow RSSI and RX voltage, the problem with 2.1 is that user needs to define which telemetry sensor represents them. I think we should have a new choice (in telemetry setup page) where user can select which sensor is RX voltage (and which is RSSI). That sensors are then allowed to be used in FAI mode by the logical switches, special funcitons, Lua scripts.

lshems, 2016-04-27:
I prefer the 2.0 way of handeling rssi and battery on radio level, auto discovered.

I see lots of people not using battery alarms anymore because they simply don't set up their receivers with LS etc.

This would also sort out the fai mode thingy. Last thing we want is getting the taranis banned for competition use, and having users assign settings is no way to go.

Just my point of view.

projectkk2glider, 2016-04-27:
>I prefer the 2.0 way of handeling rssi and battery on radio level, auto discovered.

While I agree with this, the change of the telemetry system has brought us some advantages as well as disadvantages. I don't see how can we have a very open/universal telemetry system (that was introduced in 2.1) and at the same time have some fixed telemetry values like it was in 2.0. It just does not go together very well. If we don't think this trough it could become a mess. I agree that setting up the battery alarms now is more cumbersome, but again I see no easy way out. Do you?

> and having users assign settings is no way to go.

You are presuming that A1 or A2 is used for receiver voltage. That may be true in most case, but somebody else might be using Cells or VFAS for their battery pack. So again I don't see a way how we can fix which telemetry values are allowed in FAI and which are not.

If somebody really wanted to cheat it (already) has several possibilities.

fraternl, 2016-06-01:
Why don't you just scrap that whole FAI thing?
The concept doesn't work with open source if at all for obvious reasons....

projectkk2glider, 2016-06-02:
I personally have no experience with FAI officials and have never used the FAI mode myself. It is also true that individual could *engineer* the OpenTX source code to circumvent the FAI rules while still presenting the radio as FAI compliant. But that would be cheating. 

My thinking is to provide a working FAI mode to the honest users.

agibson2, 2018-02-27:
I just ran across this same issue at the last contest I went to when I enabled FAI Choice mode on my Horus.  Normally I just unplug the vario/altimeter but the G-RX8 has a vario/altimeter built in to it.  I figured... no problem... I can finally use the FAI choice feature.  Unfortanately I had to live with not getting RSSI or low battery warnings with it enabled.  I also had to remove any value Widgets that showed the altimeter/vario values and also my F3K Training and launch scripts which access the vario/altimeter sensor data.

There is a way to disable the vario on the G-RX8 using a the bind button on the Rx but the receivers are buried deep into my DLG models because of fuse size restrictions so I can't get down into it to enable and disable it when I need to.

An FAI option I believe would still be very useful to implement.  The argument that it can be bypassed as a reason to not implement it is just as valid as someone that cheats by hides an altimeter or even a self contained vario/altimeter into the wing or something of their model.  Cheaters can cheat if they want using hardware or software.  Features like this make it easy for people who want to abide by the rules.  It was nice that OpenTX had a feature like this and others didn't too :).

lshems, 2018-02-27:
Supporting this request, and a suggestion to fix it in the telemetry setup:

Just allow a user to tick a flag on the (discovered ) sensors, saying they are ALLOWED., It should be standard at not allowed on discovery.

When activating FAI mode, all sensors not ticked FAI are disabled, and discovery is disabled permanently until firmware reinstall.

On firmware install, both sticks are zero range calibrated. There is a clear message saying you MUST calibrate, like the no failsafe alert. Introduce an extra step in calibration procedure: calibrate throttle first, then all the rest.

This would also prevent inadvertently reinstalling firmware and for orgetting to change modes (as I have done several times).

mpjf01, 2018-02-27:
Just a comment really, for consideration. I don't believe that contest directors and other competitors in major events are going to readily accept that aircraft carrying unapproved telemetry sensors or able to be operated semi-autonomously are OK to use because of a software configuration change able to be invoked by the operator. The only option that is universally acceptable is for the hardware in the aircraft to be incapable of operating outside the rules, ever. So that means receivers like the SxR and G-RX8 would not be able to be used even though the stability functionality and vario can be switched off. 

This is not the place for a debate about the usefulness (or otherwise) of hardware with these abilities. It has been recognised here in the past that in order to comply with current FAI Rules such features must not be used. Sure, people might be able to hide sensors inside inaccessible parts of the aircraft or disguise "stability" receivers as normal models, but they won't be much use to the competitor during the contest in that state - and that doesn't alter the situation. 

This whole issue will become more significant over time as (presumably) more and more manufacturers follow the trend and embed functionality that happens to be FAI forbidden inside hardware that can be difficult to distinguish from variants without those features. There may come a time when manufacturers have to produce hardware that is certified legal for FAI events and is sealed. Alternatively, the FAI might alter the rules. 

In the meantime I would recommend that those proposing the firmware enhancements try to ascertain whether or not the end result would be acceptable to the FAI before investing in a lot of work only to discover that it is not. I would be afraid that CDs would not accept the use of this type of hardware without explicit FAI approval of the methodology.

I write as a person who has organised and run (as CD) hundreds of aeromodelling contests over 40 years including a World Championships, many national and state championships and lots more local events.

myckey, 2018-02-27:
As a competitor I agree 100% with mpjf01 above, in my experience CD's and helpers simply don't have the time, knowledge off or the inclination to check that hardware that COULD be illegal has been software disabled.

My experience is that most CD's allow self certification but such an issue is more likely to come up in a protest and then it would be up to the competitor to prove innocence, Also bear in mind that there are a lot of different sensors these days and different interpretations, for example the telemetry from a receiver bus, lost frames back up failsafes etc.

I think it is up to the competitor to prove that such illegal devices are not in use by whatever method they choose, removal (if possible) being the obvious one.

lshems, 2018-02-27:
I do agree with the comments on future developments. On the FAI part I do not agree. It is not (only) the FAI that is involved here.

Selfcertification is the scheme I propose: when activating FAI mode, all telemetry is disabled apart from those deliberately tagged as FAI allowed by the pilot.

Why is it not the FAI? Because at a lot, if not all, of European contests it IS up to the pilot to behave properly. If we would have such an FAI option, that is properly implemented, it IS sort of easy to behave properly. Never have I seen a contest for F3K where the transmitter or receiver are checked by the CD. They simply don't have the time, nor knowledge about all possible setups.

But of course, there are other ways to behave properly, and to ascertain it.

I see no reason a CD could object using a alti integrated receiver for F3K, if there is no reason to suspect it will be used. If the user activated FAI mode, it can easily be shown by entering the telemetry screen that the sensors are disabled. 

The level at which you compete does make a difference as well.

Anyway, the discussion about priority is much more relevant. And I do agree that it is not high priority. But not on the arguments put forward, in my humble opinion.

mpjf01, 2018-02-27:
I don't think what I have proposed is hard to understand. There are two basic options. Option 1 is for someone here to define a mechanism that they believe complies with the FAI rules and implement it -
 only to risk having it fail when it is rejected by a CD the first time it is tried to be used at a major contest, The second option is to submit a proposed mechanism to the FAI (CIAM) through any national aeromodelling body and receive approval to use it - likely with negotiated and agreed alterations - and then implement something that you know will be accepted in any contest up to world championships level. 

Why waste developer time implementing something that cannot be used. If the argument is that CDs never check models at contests in Europe and rely simply on the honesty of competitors then maybe, for them, no special mode is needed at all.

agibson2, 2018-02-28:
It is still useful for those that need a quick way to disable sensors even without any kind of FAI approval.  I have never gone to a national team select or an international competition and don't plan to.  I mainly compete at the local level within several hours drive from me to have fun and to fly with others.  This feature is still useful events that trust the competitors to do the right thing and disable sensors which is quite a few based on experience.  It is even useful for non FAI events where people just want to disable varios and altimeters quickly and make sure they are off for friendly non official events with a few people.  I would not call that a waste of developer time but I understand that other features are likely to have higher priority.  FrSky is releasing new hardware pretty quickly to have to keep up with development wise.

I personally would be glad to even see a simpler change that also disables all telemetry sensors even RSSI and RxBat from Lua/Widgets.  It already disables them from LF functions.  I can monitor the battery level myself and would just live without knowing if I am going out of range or not.  I do my testing and know about how far I can go.  I would certainly miss RxBat and RSSI but that is how it works now anyway so we wouldn't be any worse off than how the feature kind of functions as it is now.  The feature as it is implemented now doesn't work for any use case I can think of except people that don't have any sensor data displayed or widgets that use sensor data.  Why would someone have sensors and not have them displayed or processed though so that is not very likely.

3djc, 2018-02-28:
@agibson2  do you mean in FAI mode you still have the widgets working ? If so, I will fix that

agibson2, 2018-02-28:
Yes.  All Widgets still get sensor data in the current implementation when FAI mode is enabled.  I enabled FAI mode and my Widgets were still showing altitude.  LFs could not read the sensors but Widgets were still showing the altitude and vario values (value Widget and a customer launch height Widget).

3djc, 2018-02-28:
Yes, I have borrowed an Horus and yes, there seems to be an issue. Will be fixed

agibson2, 2018-02-28:
That would be awesome.  I had to remove all my widgets quickly before the contest that morning and set them back up after the contest when I got home and disabled FAI mode again on the computer.  This would be great if FAI mode disabled Widgets/LUA? from accessing the sensor data too.

3djc, 2018-02-28:
My intend is to disabkle ALL access to telem values (widget, telem screen, LS,  PlayVal, LUA scripts). If possible, I would like to allow data in logs for later analysis tho

agibson2, 2018-02-28:
I like it.  While it doesn't solve all of this bug report as FAI allows receiver battery and RSSI to be read, it at least makes it consistent so that the sensor data cannot be read at all.

3djc, 2018-02-28:
I may fix that while I'm at it :)

agibson2, 2018-02-28:
That would be very nice indeed!  I am excited about using this in the future.  I will probably not get to sleep tonight.  lol.

3djc, 2018-02-28:
No commitment on that yet, need to confirm feasability first on rssi/rxbt exception

agibson2, 2018-02-28:
Understood with 'may' :).  It sounds like it might not be simple to fix the RSSI and RxBat going by the info posted so far.  Any work on this is appreciated though.

schwabe, 2018-02-28:
Yeah, also keep in mind that there is no real way for OpenTX what RXBT is apart from you telling opentx that that telmetry item is rxbat. If you want to cheat you can just redefine your altitude as voltage and show it as battery indicator on the main screen. Maybe scale it so that it always reads 4,xxV where xx is your altitude.

3djc, 2018-03-01:
I have a version ready that adresses the issues mentionned in the oroginal post

- all sensors with exception of allowed sensors (see bellow) have their value set to 0 in FAI mode. LS, widgets, logs.. all  aspect of OpenTX will keep seing those sensors, but with value of 0

- the following sensors are allowed, and will not have their value 'zeroed':
FrSky Sport RSSI and RxBt
FrSky old telem RSSI and A1 (RxBt)
Corssfire RSSI1, RSSI2 and RxBt
and can therefore be used normaly in widget, LS,...

3djc, 2018-03-02:
When it gets merged, it will be available in nightly, so you will be able to try it on your radios (ON THE GROUND ! as you say it very well)

3djc, 2018-03-02:
@agibson2 should be in tomorow nightly

@tgreer would it be possible for you to do a test with crossfire (since I do not have the required gear) ? Just DL a FAI option nightly fimrware later the tomorow, check that only RSSI and RxBt are available, and put back your regular firmware. Would be much appreciated

closed 2018-03-02 by 3djc

agibson2, 2018-03-03:
I see that it was committed on the 1st so I installed the nightly from the 2nd and I still see the same behavior so I assume it is not in the nightly from the 2nd.  Is that correct?

3djc, 2018-03-03:
Yes, I coimmited on the second in european timezone, and it is built in moning, european timezone too. N535 should have it

agibson2, 2018-03-03:
I can't get N535 to write to the radio (using Horus).  I upgraded Companion to N535 before trying.  I tried about 4 times.  I tried multiple USB ports and a different cable.  It fails at the same spot each time (somewhere around the middle of the upload).  I can write N534 back to it without issue.  I assume it is just a bad binary build or something.

`C:/Program Files (x86)/OpenTX/Companion 2.2/dfu-util.exe -a 0 --dfuse-address 0x08000000:2097152 --device 0483:df11  -UC:/Users/Myuser/AppData/Local/Temp/12532-flash-check.bin
=================================================================================
dfu-util 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Filter on vendor = 0x0483 product = 0xdf11
Opening DFU capable USB device... ID 0483:df11
Run-time device DFU version 011a
Found DFU: [0483:df11] devnum=0, cfg=1, intf=0, alt=0, name="@Internal Flash  /0x08000000/04*016Kg,01*064Kg,07*128Kg,04*016Kg,01*064Kg,07*128Kg"
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
bytes_per_hash=2048
Starting upload: [###############################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################] finished!

=================================================================================
C:/Program Files (x86)/OpenTX/Companion 2.2/dfu-util.exe -a 0 --dfuse-address 0x08000000 --device 0483:df11  -DC:/Users/Myuser/Documents/OpenTx22/Firmware/opentx-x12s-faichoice-lua-en-2.2.2N535.bin
=================================================================================
dfu-util 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Filter on vendor = 0x0483 product = 0xdf11
Opening DFU capable USB device... ID 0483:df11
Run-time device DFU version 011a
Found DFU: [0483:df11] devnum=0, cfg=1, intf=0, alt=0, name="@Internal Flash  /0x08000000/04*016Kg,01*064Kg,07*128Kg,04*016Kg,01*064Kg,07*128Kg"
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuUPLOAD-IDLE, status = 0
aborting previous incomplete transfer
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 1449632
............................................................................................................................................................................................................................................................................................................................................................................................................................No valid DFU suffix signature
Warning: File has no DFU suffix
dfuse_download: libusb_control_transfer returned -9
Error during download
Failed to write whole chunk: -9 of 2048 bytes

=================================================================================`

3djc, 2018-03-03:
works here, with firmware downloaded using companion (not compiled here, so exactly as you). Maybe you antivirus playing funny or something like that ?

C:/Program Files (x86)/OpenTX/Companion 2.2/dfu-util.exe -a 0 --dfuse-address 0x08000000:2097152 --device 0483:df11  -UC:/Users/Jean-Christophe/AppData/Local/Temp/9548-flash-check.bin
=================================================================================
dfu-util 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Filter on vendor = 0x0483 product = 0xdf11
Opening DFU capable USB device... ID 0483:df11
Run-time device DFU version 011a
Found DFU: [0483:df11] devnum=0, cfg=1, intf=0, alt=0, name="@Internal Flash  /0x08000000/04*016Kg,01*064Kg,07*128Kg,04*016Kg,01*064Kg,07*128Kg"
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
bytes_per_hash=2048
Starting upload: [###############################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################################] finished!

=================================================================================
C:/Program Files (x86)/OpenTX/Companion 2.2/dfu-util.exe -a 0 --dfuse-address 0x08000000 --device 0483:df11  -DC:/Users/Jean-Christophe/Desktop/opentx-x12s-faichoice-lua-en.bin
=================================================================================
dfu-util 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Filter on vendor = 0x0483 product = 0xdf11
Opening DFU capable USB device... ID 0483:df11
Run-time device DFU version 011a
Found DFU: [0483:df11] devnum=0, cfg=1, intf=0, alt=0, name="@Internal Flash  /0x08000000/04*016Kg,01*064Kg,07*128Kg,04*016Kg,01*064Kg,07*128Kg"
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuUPLOAD-IDLE, status = 0
aborting previous incomplete transfer
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 1449632
....................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
File downloaded successfully
No valid DFU suffix signature
Warning: File has no DFU suffix

=================================================================================

3djc, 2018-03-03:
Using this site: https://defuse.ca/checksums.htm
And checking the firmware I get:
md5 | 6122530d17b16cbdb78904b86107214a

agibson2, 2018-03-03:
Thanks for testing it.  I deleted the firmware and force another download and it worked.  The file sizes were the same but obviously something inside was corrupt on the first download.

3djc, 2018-03-03:
ok, cool

agibson2, 2018-03-03:
It looks good so far.  I have tested my plane as best as I can.  RxBat and RSSI are available but vario and altimeter are read as 0.  Awesome stuff!  Liking it :).

I noticed that when you read your radio settings when FAI choice is enabled, the screen that comes up might might not be accurate anymore as it mentions the telemetry screen is unavailable?  Not a big deal at all of course :)

`OpenTX Companion
If you enable FAI, you loose the vario, the play functions, the telemetry screen.
This function cannot be disabled by the readio.
Are you user ?`

I am not sure if you want to change the wording or not.

agibson2, 2018-10-28:
Has anyone had trouble unlocking FAI mode?  I enabled the feature.  It disabled things properly but now it is stuck in 'Locked in FAI Mode'.  I uploaded the same firmware and it still shows the same status.  Am I missing something to unlock the FAI mode that I just forgot or is this broken in 2.2.2?

PR #5736 (2018-03-02): FAI mode reworks
all sensors with exception of allowed sensors (see bellow) have their value set to 0 in FAI mode. LS, widgets, logs.. all aspect of OpenTX will keep seing those sensors, but with value of 0

the following sensors are allowed, and will not have their value 'zeroed':
FrSky Sport RSSI and RxBt
FrSky old telem RSSI and A1 (RxBt)
Crossfire RSSI1, RSSI2 and RxBt
and can therefore be used normaly in widget, LS,..

This fixes #3443 and more