opentx/opentx #3445: Companion Curve Editor, improve visualisation of curve points

Spainman, 2016-04-26:
Two suggestions that I personally think would improve usability and shouldn't require much changes code wise (assuming the plotting library doesn't suck) + a bug:

**Possible improvements:**
a) Let's get rid of the shadows under the curve points

  - personally I think they fail as 'eye candy' 
  - they hinder the visualization of the actual point position (I get an impression of double vision and no - there is no drunken OpenTx-ing involved ;-) )

b) Minimize the size of the curve points 

  - personally I operate on the table values (works great with the keyboard cursor keys + tab, shift tab) and not with the mouse pointer, so a much smaller size would provide a better visualization of the transitions on the points
  
- maybe the graphing framework even provides the option to set the focus range different from the visual size, so that handling the points with the mouse pointer would not have to be impacted?

- if there is no such option, I still think that a point size that allows a good handling with the mouse but still is significantly smaller than it is now would still be an improvement

Opinions?

**Bug / missing functionality (?) :**
c) The smoothing option does not visually smooth the curve when editing the points manually (= not using the curve generator)
  - I'm not sure it is intended. I usually edit the curves on the transmitter and there the smoothing obviously works when editing the points (which is lovely and a much appreciated functionality btw :+1: )

projectkk2glider, 2016-04-27:
Related: #2998, #1911 

This issue is actually a duplicate of #1330, I will close that one, since this has more info.

mpaperno, 2017-06-03:
Related: #4972

closed 2020-04-06 by stale[bot]