opentx/opentx #3447: Changing MAX to USE

Oldgibber, 2016-04-27:
Brought a concern up about MAX up some time ago, but it seems to have been lost with the plethora of other work, and so have closed it down. However, this suggestion seems a very simple fix which makes things more logical.

There was general agreement that MAX was not ideal, but then having MAX/MIN or FIX or SET were not ideal either. How about USE?

USE Weight(+100%) Switch(SA↑)
USE Weight (0) Switch(SA-)
USE Weight(-100%) Switch(SA↓)

It would appear to be a very simple programming change, and if, as I suppose, MAX is simply a stored code number in the eepe file then on future versions it would still be correct but with USE instead of MAX

hrenz, 2016-04-28:
Were is your Problem, cant see any
Max is Max 
and Max is +100%
you can tell it mickymouse but its +100%

Oldgibber, 2016-04-28:
Its simply not intuative. Its clumsy. 

How does a new user know MAX means +100?

tomkfly, 2016-04-28:
I don't see how USE is any more intuitive.

kilrah, 2016-04-29:
> How does a new user know MAX means +100?

The same way they's know USE is +100% since it's no more intuitive - either by reading docs or trying it.

And given changing it would render years of documentation efforts obsolete it makes absolutely no sense to change it "just to change it", which is why the old issue would have been closed... and this one is too.

closed 2016-04-29 by kilrah