opentx/opentx #3453: Taranis Plus Hardware Settings: Please generalize LS and RS

Spainman, 2016-04-29:
Currently the firmware does not allow customizing LS and RS in regard of the pot type. 

 - I would like to request the option to select 'Pot' (= pot without detent) for any potentiometer, including LS/RS, in order to allow for modifications that involve centerless sliders. I know that the selection only impacts the the calibration/linearization and that the ADC monitor helps centering the input before calibration but that's not for everyone.
 
 - Maybe even add 'Multipos Switch'? I don't know how involved that would be software wise. But having a rotating input with more than a hand full of discrete positions available right at the index finger without having to let go the sticks seems great for things like flaps deployment, camber presets etc. 

Question on the topic regarding the Taranis Plus hardware and pot/slider inputs (= excluding the sticks) in general:
Are all adc-inputs opamped in the same way or differently in order to compensate for the different throw of LS/RS and S1/S2? I ask because I'm thinking of replacing RS with a 270° pot and a wheel type actuator. Thank you.

projectkk2glider, 2016-04-29:
Regarding the second part:

All analog inputs are feed trough a RC filter (220 ohm and 100nF) directly to the CPU pins for ADC multiplexer. There are no op-amps in between. Therefore take care:
 * not to exceed 0V - 3V voltage on any analog input
 * avoid increase in pot resistance as this will increase effective ADC input impedance and therefore increase ADC cross-talk between ADC channels. It is best to measure existing pots and use the same value.
 * OpenTX can calibrate any value from 0 to 3V so there should not be any problem. But if you use for example 270 degree pot and only have usable movement of 90 degrees, then the ADC resolution will be decreased. But the full range of -100% to 100% will still work in the OpenTX after the calibration.

projectkk2glider, 2016-04-29:
Other suggestions will require an EEPROM change. I don't see any reason why this should not be implemented. Let's see what other devs have to say.

bsongis, 2016-04-29:
Just time missing for me, be careful with the conversions if you start working on this one!

Spainman, 2016-04-30:
@projectkk2glider:
Please allow me to extend my question although it's a bit beyond the scope of this ticket: When you say that _all_ analog inputs are non-amplified, do you mean the sticks too? I've seen some discussions in different forum threads that imply that the stick pots are amplified, so I'm a bit lost right now. Thank you

projectkk2glider, 2016-04-30:
True, some gimbals gave amplifiers built in, but the connection to the main board and CPU is still the same, 0 - 3V.

Spainman, 2016-04-30:
So on the mainboard itself are no amplifiers involved at all? So in case of the taranis plus (no amp boards in the stick assembly itself) there is simply no amplification whatsoever?

kilrah, 2016-04-30:
None of the FrSky radios have any gain control on any analog channel on the main board.

Early Taranis gimbals had amplifiers on the stick until they sourced appropriate pots, but since then all the pots simply have the correct electrical range for their purpose. Gimbals and sliders are 60° or so, face pots are standard 270°.

Spainman, 2016-04-30:
Thank you for the confirmation.
Regarding the stick pots: You are right! The pot's look and smell like regular Murata SV series (even from the inside - the visible resistive track looks the same as the 330° version) but they aren't. Just took a measurement and they max/min out at around +/-30° of rotation and are ~2.3k instead of 10k. Wouldn't have thought that...! Nothing to improve here...

closed 2020-04-06 by stale[bot]

arikrupnik, 2021-06-28:
Is there any reason not to treat slider and pots the same?

I recently upgraded to a TX16S. The sliders have weak detents and the readings fluctuate around zero. I wish I could mark them as "pots with detent." and solve the problem that way. What is the difference between pots and sliders anyway?