opentx/opentx #3456: Telemetry sensors with number 20 or higher are not available in lua scrips

pau33, 2016-05-02:
I wrote a lua script (/mixes) to do some calculations with Telemetry Sensors but did not get all sensors as an input. I looked a little deeper an found out, that Telemetry Sensors with a number of 20 or higher were not choosable as Input on the lua custom script page.
Maybe it is an issue with hex and dec, if you test on the highest possible sensor number.
Can you confirm this behavior?
Probably this could be easily fixed in the future.
Taranis plus and opentx 2.1.8.

Regards, Paul

projectkk2glider, 2016-05-02:
Can you post a short example script. Thanks.

projectkk2glider, 2016-05-02:
@bsongis we have a problem. Script input value is `int8_t` which is not enough for `MIXSRC_LAST_TELEM`:
```c
PACK(typedef struct {
  char    file[LEN_SCRIPT_FILENAME];
  char    name[LEN_SCRIPT_NAME];
  int8_t  inputs[MAX_SCRIPT_INPUTS];
}) ScriptData;
```

bsongis, 2016-05-02:
Yes. But I don't think we should change it in 'master' now. We have the problem on the X9E as well, not only in Lua scripts.

projectkk2glider, 2016-05-02:
Sure, it will require EEPROM change. Unless we can come up with the workaround. The type field in `struct ScriptInput` has 7 bits free.

bsongis, 2016-05-02:
But this struct is not in EEPROM!

bsongis, 2016-05-02:
In 2.2 I propose to change the ScriptData structure this way:
- only 6 inputs instead of 8: we save 2 bytes
- only 6 chars for filename (as all other filenames in 2.2) : we save 2 bytes
- only 6 chars for name : we save 2 bytes
- 16 bits for each input : we use the 6 spared bytes

projectkk2glider, 2016-05-02:
I agree.

bsongis, 2016-05-02:
We can't fix this problem in 2.1 as we don't want to change the EEPROM structure

closed 2016-05-02 by bsongis

projectkk2glider, 2016-05-02:
@pau33 this will work in OpenTX 2.2

pau33, 2016-05-02:
That's okay, Tank you
Paul

commit afce1744 (2016-05-02): Fixes #3456