opentx/opentx #3457: Trims to Subtrims problem

Dfrazier1965, 2016-05-02:
I have a Taranis Plus on 2.1.8 and I have a small issue. It was brought to my attention by my brother-in-law and I verified it on mine also.
 Using trims to subtrims at the bottom of the outputs page, applies the subtrims necessary and centers all trims as expected. When on a single channel, long pressing on Enter brings up a choice to apply trim to subtrim for that channel, but using this method does not center the trim. Is this how it is intended to work or is it a bug?

projectkk2glider, 2016-05-02:
I would say it is done like this on purpose. Imagine situation where one trim affects two or more channels.

fraternl, 2016-05-02:
Ah... I didn't think of that... (I encouraged him to create the ticket).

bsongis, 2016-05-02:
Yes exactly, @projectkk2glider is right and this is done on purpose!

closed 2016-05-02 by bsongis

Dfrazier1965, 2016-05-02:
Didn't think of that. Makes perfect sense.