opentx/opentx #3459: Flight modes rotary encoders linked flight mode plus one causes Companion to freeze

elecpower, 2016-05-04:
Setting the linked flight mode for a RE to one FM number greater than the current flight mode causes the application to freeze. There is no freeze if the current FM is the last FM since there can be no higher to link to. 
Companion 2.1.8 Ubuntu 14.04

kilrah, 2016-05-04:
What do you mean "linked flight mode"?
What radio type?
Can you give a configuration example?

elecpower, 2016-05-04:
Any of the 9X radios with rotary encoder support eg Turnigy 9XR PRO
Appears to be caused by a relative pointer. To test set each RE own value to the FM number. Then change to point to another FM. In the screen shot the combo box has FM 8 but the value is from FM 7 value. 
[image]

kilrah, 2016-05-05:
Can you check if that's not related to the refresh issue? I.e. change the combo, close the window then reopen it, the value likely is updated now.

elecpower, 2016-05-05:
Does not appear to be related to refresh. I believe I have found the errant code and have written a fix for Companion which seems to fix the issue but I need to do some more testing. If my fix works I also need to locate the matching radio model edit code to determine if the same coding logic bug exists.

commit 5eca36b9 (2016-05-14): Fixes #3459 Rotary encoders linked flight mode can freeze Companion (#3470)
commit 98798c1e (2016-05-14): Fixes #3459 Rotary encoders linked flight mode can freeze Companion (#3470)
closed 2016-05-14 by bsongis