opentx/opentx #3461: [Enhancement] S.Port sensor configuration from the radio

pawelsky, 2016-05-05:
As it is already possible to upload firmware to TX/RX via the S.Port, how about extending this functionality and adding an option to configure S.Port sensors through the same interface?

In the minimal version one could change the sensor ID (to avoid conflicts when e.g. two FLVSS sensors are used) in the more capable version this could also include changing configuration of sensors (e.g. SP2UART baudrate/mode or  Vario/FLVSS/FCS data rate).

Currently multiple tools/devices are required to do this ([S.Port Tool](http://www.frsky-rc.com/download/view.php?sort=&down=143&file=S.Port%20Tool), MikeB's [SportSet](http://openrcforums.com/forum/viewtopic.php?p=78385&sid=ae4b7a074c8ad623f3efb3858499e2f8#p78385), FrSky's [SCC](http://www.frsky-rc.com/product/pro.php?pro_id=73). It would be nice to have all this functionality in one place - the radio.

schwabe, 2016-05-06:
Hm, for none of the tools have the source code available but SportSet is from @MikeBland. @MikeBland   any chance that you have implemented that in ersky9x or have the source code available of your tool? 

Or alternatively is the protocol used to configure the sensors available anywhere?

pawelsky, 2016-05-06:
Yes, the protocol is described in "Smart port protocol V3.3.2.pdf" which can be obtained from FrSky after joining their [Developer Union](http://www.frsky-rc.com/BBS/viewtopic.php?f=20&t=125)

fraternl, 2016-06-14:
Duplicate of a ticket I created a year ago
https://github.com/opentx/opentx/issues/2001

fraternl, 2016-06-15:
I know you have pragmatic reasons to close my ticket and not this one, but it still hurts (mildly)  ;-)

MikeBland, 2016-12-13:
Hmm, lack of time for everything!
Actually the original code for the UART in SportSet is also in eepskye, and that is on Github already, so is available.
I will be putting SportSet sources up on Github, just need time to do so.
The current version now also supports the S6R configuration (via a STK) and also configuration of the RPM sensor (number of poles).
I did start on putting some of this into the radio using "maintenance mode" in ersky9x.
Since the latest openTx supports sending SPort data from a LUA script, maybe the other options required could be done that way, or will normal SPort operation cause a problem?

pawelsky, 2016-12-13:
LUA script in 2.2 is indeed an interesting suggestion. That would be much more flexible approach...

closed 2020-04-06 by stale[bot]