opentx/opentx #3462: Simulator crashes with lua script

Spainman, 2016-05-05:
- create a lua script with playTone(2000, 1000, 20, 0, 100)
- companion and stand-alone simulator crash after running the playback 
- tested on Windows in Companion 2.1.8 running a function script and also as a one time script 

In the meantime I would appreciate if somebody could help on how to use playTone. Reference guide lacks useful information on the _pause_ argument. Actually I was testing around with values to discover what pause might mean... Thanks.

Spainman, 2016-05-06:
Thank you for the bugfix. Still I would appreciate if you could comment on the parameter _freqIncr_. What are non-abnormal parameter values?

projectkk2glider, 2016-05-06:
Perhaps you can help me formulate the text for *freqIncr* for Lua docs. Here is how it works:
* tone starts playing with the frequency that was set by the first parameter *frequency* (unit is Hz)
* every 10ms frequency is changed with a formula:
  * newFrequency = oldFrequency + 10 * freqIncr

Couple of examples:
 * `playTone(1000, 1000, 0, 1)` starts with 1000Hz, increase by 10Hz every 10ms, at the end (1s play time) the frequency reaches 10Hz * (1000ms/10ms) + 1000Hz = 2kHz
 * `playTone(1000, 100, 0, -5)` starts with 1000Hz, decrease by 50Hz every 10ms, at the end (0.1s play time) the frequency reaches -50Hz * (100ms/10ms) + 1000Hz = 500Hz

The valid range for *freqIncr* is from -127 to 127. Even small number has big effect.

closed 2016-05-06 by bsongis

Spainman, 2016-05-07:
>Perhaps you can help me formulate the text for freqIncr for Lua docs

Thank you for explaining!!! See my post in the issue where we are discussing lua mix script failsafes. I will try to help with the guide. For now I would suggest you just add the description you just wrote here. I will then try to improve on it right in the guide. Ok?

PR #3468 (2016-05-06): Fixes #3462: PlayTone check for min/max frequency added. Fixes Lua pl…
…ayTone() crash when called with abnormal parameter values.

Closes #3462
commit 2e631f99 (2016-05-06): Fixes #3462: PlayTone check for min/max frequency added. Fixes Lua playTone() crash when called with abnormal parameter values. (#3468)
commit 6349ff94 (2016-05-06): Fixes #3462: PlayTone check for min/max frequency added. Fixes Lua playTone() crash when called with abnormal parameter values. (#3468) (ported from master)