opentx/opentx #3463: Taranis reporting incredible climb speed

Julez1, 2016-05-05:
Hey guys,
I took a plane to the airfield today which I had not flown for half a year, and never flew on my new Taranis Plus, only on my old non-plus Taranis. I noticed a strange bug with my high precision vario's climb speed. As soon as the climb speed is slightly negative, it reports a positive valure close to 1/3rd of the speed of light. I made a screenshot of that problem. Before I apply for a job at SpaceX for obvious reasons, could you have a look at it? The model in the eeprom is the Elektro-Master or E-master, forgot how I named it. Other models did not show that issue so far.

Thanks,
Julez

[image]

[Standard 56.zip](https://github.com/opentx/opentx/files/251096/Standard.56.zip)

projectkk2glider, 2016-05-05:
OpenTX version 2.1.?

Julez1, 2016-05-06:
Yes, Taranis and Vario both have the newest firmwares.

projectkk2glider, 2016-05-06:
Did you have vertical speed logged onto SD card? If yes, are numbers there also weird?

Oldgibber, 2016-05-06:
I've found the same. This is with an X4R
[image]
Just these two weird numbers. Everything else within a range of ±10.

projectkk2glider, 2016-05-06:
@Julez1 are all negative values bad or only occasionally?

I don't use FrSky vario myself, I use OpenXsensor with D8 (hub protocol) and X4 (SPORT protocol) receivers and I occasionally (very rarely) get a sporadic value for some sensor (sometimes altitude, sometimes vertical speed). 

There are a couple of possible reasons why this is happening:
* the sensor actually sends such value - a bug in sensor
* the data gets corrupted in transport (sensor - receiver - radio TX module - main CPU) but the relatively simple CRC check does not catch it.
* some other bug in OpenTX code that handles telemetry

I any case this is a hard issue to debug. More data would help.

Julez1, 2016-05-06:
All negative values are bad. As soon as it is negative, it writes the high number. Just captured a log. It also shows this has nothing to do with RSSI.
[E-Master-2016-05-06.zip](https://github.com/opentx/opentx/files/252883/E-Master-2016-05-06.zip)
[image]

projectkk2glider, 2016-05-06:
That is really strange. I presume your setup worked OK on old radio (with older OpenTX version). I find it strange that there are not more users with the same problem if it is so obvious.

The values you get are quite interesting in hexadecimal:
* 1073741822 == 3FFFFFFE
* 1073741823 == 3FFFFFFF

We first need to establish it this is a problem with your sensor/radio or general OpenTX problem. Can you try changing one components one by one in order to determine if it is a particular sensor or not.

projectkk2glider, 2016-05-06:
@Julez1 are you using D or X series receiver in this model?

Julez1, 2016-05-06:
The issue appears to be model memory specific.
I can apparantly change componants as I like with the problem persisting. In the affected E-Master plane, I have an X8R and an old 100A current sensor using the vario as a hub. Disconnecting the hub from the vario or connecting another vario made no difference. Also another X6R with the other vario had the same error.
But on another model memory, the Phoenix, I did not have the problem. Not with the Phoenix plane itself and not with the X6R test reciever with the vario as mentioned above. 
So all components connected when the E-Master memory was active had the error, and no components connected with the Phoenix memory being active had problems.

projectkk2glider, 2016-05-06:
I think I found the bug, you must have `filter` enabled, please try disabling it.

Julez1, 2016-05-06:
Yes indeed. After disabling the filter, everything works normal and as expected. Thanks, man!

projectkk2glider, 2016-05-06:
Ok then, I will fix this bug, in the meantime you have a workaround :smile:

Julez1, 2016-05-06:
That was fast. Thanks for your work.

PR #3469 (2016-05-06): Projectkk2glider/issue 3463 negative filtered sensor values
Closes #3463

Please check out the second commit, I think it was a bug.
commit 6630406d (2016-05-06): Projectkk2glider/issue 3463 negative filtered sensor values (#3469)

* Fixes #3463: negative sensor values were invalid if filter was active for the sensor

* Re #3463: enable simultaneous use of `offset` and `filter` (was: when `offset` was enabled, the `filter` was always inactive even if enabled)
commit 8cd677d6 (2016-05-06): Projectkk2glider/issue 3463 negative filtered sensor values (#3469) (ported from master)

* Fixes #3463: negative sensor values were invalid if filter was active for the sensor

* Re #3463: enable simultaneous use of `offset` and `filter` (was: when `offset` was enabled, the `filter` was always inactive even if enabled)
closed 2016-05-06 by bsongis