opentx/opentx #3464: Feasibility

perlcat, 2016-05-05:
I really like the opentx project. I looked at the list of issues, but didn't see anything about an idea I had. I have a raspberry pi, and I have a really nice gamer joystick. Would it be possible to write something on the pi that would allow it to use my joystick over the trainer port on my transmitter? I can do stuff with SDL and perl, but it probably isn't the best place for that language.  Before I went to the effort of doing this, I thought I'd see if you guys thought it was feasible and worth the effort.

lshems, 2016-05-06:
Why not use the joystick function on the USB. Someone made it work for an multibutton joystick just some time ago.

perlcat, 2016-05-06:
Maybe I don't understand USB correctly, but the connector on the back of my
X9D is a mini -- which does not provide power. Is there a USB connector
inside somewhere that provides power?

kilrah, 2016-05-06:
Lshems is referring to the function allowing the Taranis to BE a joystick and didn't understand it's the opposite you're after.

Generating PPM from a Pi might be a pain, but you should be pretty easily able to put out SBUS and simply use the Taranis' trainer function as that's just data on a serial port.

closed 2016-05-06 by kilrah

lshems, 2016-05-06:
Yep, mixed things up.

darvin, 2020-08-01:
up

darvin, 2020-08-01:
why this extremely important feature is neglected? everyone loves ridiculous RC sticks, hates proper usb joysticks?