opentx/opentx #3465: S-Port Mirror of Taranis X9D does not send readable telemetry data

tedbarth, 2016-05-06:
I'm trying to interpret the Smart.Port data sent out over UART by my Taranis x9d Plus (with X4R receiver) when set to **S-Port Mirror** in the hardware settings dialog. I use a HC-06 Bluetooth module configured to 57600 Baud and several different  Bluetooth dongles on my PC as well as an Android device to connect to that Bluetooth Serial Port.

I am able to connect to the Bluetooth serial port and to see a reflection in data change when enabling telemetry on my RC model (I guess RSSI and other sensors). My problem is that the data I receive on that port does nothing have in common with the Smart Port protocol I can read off the OpenTX code. Not even the START_STOP byte can be found like mentioned in the following snippet and in other open source libraries on GitHub: https://github.com/opentx/opentx/blob/master/radio/src/telemetry/frsky.h#L478

What I get instead is this:

`a0 a0 e6 57 7f b8 be a0 e6 d7 f9 ff e5 a0 e6 77 
 a8 ff ff a0 a0 a0 a0 91 97 15 ff ff b5 a0 a0 a0 
 a0 a0 a0 a0 a0 a0 a0 a0 a0 a0 e6 57 7f 74 df a0 
 e6 d7 58 e5 ff f5 a0 e6 77 a8 ff ff a0 a0 a0 a0 
 a0 91 97 15 ff ff b5 a0 a0 a0 a0 a0 a0 a0 a0 a0 
 a0 a0 a0 e6 37 fb 19 a0 e6 57 7f ff a0 e6 d7 57 
 30 ff e7 a0 e6 77 a8 ff ff a0 a0 a0 a0 a0 91 97 
 15 ff ff b5 a0 a0 a0 a0 a0 a0 a0 a0 a0 a0 a0 a0 
 e6 57 7f 73 df a0 e6 d7 ab e5 ff fb a0 e6 77 a8 
 ff ff a0 a0 a0 a0 a0 91 97 15 ff ff b5 a0 a0 a0 
 a0 a0 a0 a0 a0 a0 a0 a0 a0 e6 57 dd fb ff a0 e6 
 d7 fe ff e9 a0 e6 77 a8 ff ff a0 a0 a0 a0 a0 91 
 97 15 ff ff b5 a0 a0 a0 a0 a0 a0 a0 a0 a0 a0 a0 
 a0 e6 57 7f b8 be a0 e6 d7 f9 ff a0 e6 77 a8 ff 
 ff a0 a0 a0 a0 a0 91 97 c5 ff ff a0 a0 a0 a0 a0 
 a0 a0 a0 a0 a0 a0 a0 e6 57 7f 74 df a0 e6 d7 5b 
 31 ff a0 e6 77 a8 ff ff a0 a0 a0 a0 a0 91 97 c1 
 ff ff a0 a0 a0 a0 a0 a0 a0 a0 a0 a0 a0 a0 e6 57 
 7f 73 7d a0 e6 d7 56 30 ff a0 e6 77 a8 ff ff a0 
 a0 a0 a0 a0 a0 91 97 15 ff ff a0 a0 a0 a0 a0 a0 
 a0 a0 a0 a0 a0 a0 e6 57 db fb a0 e6 d7 ac 97 e7 
 a0 e6 77 a8 ff a0 a0 a0 a0 91 97 15 ff b5 a0 a0`

As you can see, there are clearly data packages received. They are surrounded by several 0xa0 bytes (linefeed). The packages start with an 0xe6 instead of an 0x7e and the other bytes don't reflect any known control bytes, too. I hope this simply is a configuration error and not a change in a protocol or something like that.

Any idea whats wrong?

I'm using the following European (German) version of OpenTX and tranceiver module from the frsky website:

FW: opentx-tranis-plus
VERS: 2.1.7
DATE: 2016-01-05
TIME: 03:18:23
EEPR: 217

projectkk2glider, 2016-05-06:
My first idea: the serial signal needs to be inverted. Have you read https://github.com/opentx/opentx/wiki/Taranis-serial-port

hrenz, 2016-05-06:
would it be possible to send non invertetd data  at Taranis UART? simplifiedy lots

kilrah, 2016-05-06:
No.

And yes that looks like you simply have the data line inverted.

closed 2016-05-06 by kilrah

projectkk2glider, 2016-05-06:
The radio has inverter (with transistors) after the CPU output. Perhaps there is a chance that CPU UART supports inverted output mode, but this needs to be investigated.

kilrah, 2016-05-06:
Can already tell you no. Would have saved lots of issues...
Only newer generations (F3, F7) do.

projectkk2glider, 2016-05-06:
After the search I determined that the STM32 CPU does not support inverted UART.

tedbarth, 2016-05-06:
It's working after inverting the signal! Inverting the received bytes was not enough since the stop bit got inverted too. Thanks for your help!

https://github.com/opentx/opentx/wiki/Taranis-serial-port
The S.Port Mirror is mentioned as output. Is there any thing happening when sending data on the RXD toward the Taranis?

kilrah, 2016-05-06:
> Is there any thing happening when sending data on the RXD toward the Taranis?

No.

hrenz, 2016-05-06:
ok,  we need a real bit invers,  Hard or Soft solution

RexBelli, 2016-08-02:
@tedbarth Can you share what hardware you used to invert the signal?

3djc, 2016-08-02:
An Internet search of "frsky inverter"  will lead you to many solutions,  depending on how much you want to build or buy.

RexBelli, 2016-08-02:
@3djc I've been looking into those, but solid information is hard to come by.  Is the inverter for Telemetry and SBUS the same?

3djc, 2016-08-02:
An inverter is an inverter, you can have it invert what you need to. Given you have given no detail arround what you want to achieve, it is difficult to say. Also, this is not an help board, but a bug tracking system, please use a forum like http://www.rcgroups.com/forums/showthread.php?t=2351863 to get some more help and discussion going on :)

RexBelli, 2016-08-02:
@3djc Thank you :)  I would've sent a private message if github allowed it.  For reference, I was looking for an inverter to do exactly the same thing as what tedbarth used it for. (Taranis X9D+ -> inverter -> HC-06 BT module -> Android/Windows)

davidngrc, 2018-06-14:
I am trying to connect the X9D+ serial port to HC-06 bluetooth as well.
and then use "frsky dashboard" app to see the gps map.

@RexBelli
Taranis X9D+ -> inverter -> HC-06 BT module -> Android/Windows
there is a tx and rx serial, is that mean we need two inverter? one for rx serial, and one for tx serial? (I got 1 sport inverter )

I remember somewhere said something like
https://www.rcgroups.com/forums/showpost.php?p=22186856&postcount=103
x9d is a serial, and HC-06 is TTL, you need a serial rs232 to TTL converter. ( I got MAX3232CSE)
is that the same thing as an inverter ?

could anyone draw some diagram or text explaining how the pin is connected between x9d+ serial port to (inverter  or serial2ttl) then to bluetooth ?

thank you very much.

RexBelli, 2018-06-14:
Hey @davidngrc this was almost 2 years ago, so my memory is a little rusty.  For my use case, I only needed to send data from the Taranis to my computer, so I only needed 1 data line and therefore only 1 inverter.  I never approached bi-communication, so I can't say how data from computer to the Taranis would work. 

BTW, this might help with your serial vs TTL vs rs232 question (TTLS and rs232 are serial): https://www.sparkfun.com/tutorials/215

schwabe, 2018-06-14:
Bidrectional communiction via sport mirror is not imlemented I think. You only a mirror of that conection.to read for that option.

davidngrc, 2018-06-15:
For anyone interested in doing this.

My hardware: X9D+, R9 Mini (sbus+sport), Matek F405-Wing FC, HC-06 (or HC-05) Bluetooth 5600 baud rate 8N1, BN-880 GPS
The software: iNav 1.91, OpenTx 2.2.1

Wiring: X9D+ serial port TX to sbus invertor to HC-06 RX
(serial port output voltage is battery voltage, you need a BEC to get 5v or 3.3v for your bluetooth)
(no need to connect the X9D+ serial port RX and the HC-06 TX )

Android App Test
EZ GUI v4.5.450 voltage display correct, GPS does not..
FrSky Dashboard v3.0B06 do not display anything, the Bad package is increase...
FS_Logger Base v3.2.1   not working.

Any other app that works?