opentx/opentx #3466: Taranis won't bind.

3000gt, 2016-05-06:
I am having a big problem getting my Taranis to bind after I updated my firmware. It started when I bought a X4Rsb receiver and could not get it to bind because it had new EU LBT programing. Someone told me I had to go update the firm ware. I did that, and still could not get it to work. I still could fly my other models at the time but just could not get the X4Rsb to bind. i tried various other things, and now my Taranis won't bind to anything. I even did a factory reset and can't get it to bind. When I look on the SD card, there is no firmware to be seen? I have version 2.1.8 opentx-taranis-plus on my radio Any help would be great, I think I just need to put the firmware back on the SD card, but can't find the .hex files.

kilrah, 2016-05-06:
Not an OpenTX issue, what the person was telling you is that you need to match the firmware of the RF module and receivers. Tons of info about that on forums.

closed 2016-05-06 by kilrah

3000gt, 2016-05-06:
I've tried all that. I need to find the files that go on the SDcard. I've updated ALL my firmware, but there is no file on SD card.

3000gt, 2016-05-06:
Yeah, that's what everyone says,"Oh, just read a forum? They have tons of info" then they pass me on to the next guy without helping.
Thanks, I already know about all the forums and how useless they are to most people. 
Thanks anyway. I'll find someone who can, or wants to help.

kilrah, 2016-05-06:
A quick Google search brings up a good enough tutorial for what I assume to be your need as per your incomplete description, i.e. the need for flashing an X4R-SB to non EU firmware:
http://www.dronetrest.com/t/how-to-upgrade-your-frsky-receiver-firmware-x8r-x4r-x4r-sb/1667

The firmwares for the modules and receivers (.frk files) are at the obvious place - the appropriate product's page in the "Downloads" section of FrSky's website.

3000gt, 2016-05-06:
I am having a problem getting my Taranis to bind after I updated the firmware. It will not bind to anything now!!! I accidentally erased the File of the SD card, and now I can't find it anywhere? I tried even using OpenTX and re flashed my Taranis with the latest firmware, but still no luck!!! It's been a week now of me searching the internet and watching videos and people telling me how EASY it is , and that I only need to update the Taranis firmware. Well I've tried all that, and using OpenTX and it still won't bind!!! When I go to the FRsky site, the files are a .rar exstension and I can't put that on the SD card? Where do I go to get the .frk files?? I've looked everywhere and am getting ready to loose it, and just go back to Spektrum :(

schwabe, 2016-05-06:
the rar files are archives, just a program like winrar to extract them.

3000gt, 2016-05-06:
Yeah, I finally figured that out. It still won't work. It says not a valid firmware file.

kilrah, 2016-05-06:
WHERE does it say this? Please describe exactly what you're doing. Can't help you without that.