opentx/opentx #3467: [LUA] Mix scripts enhancement - failure / default output values

Spainman, 2016-05-06:
Currently when a mix script is terminated for whatever reason, every output of the deceased script is replaced with the value **0**.  Depending on the purpose of the script it can be more suitable to use a value different from 0 if the script dies (e.g. kill-switch scenarios, values that influence spoilers, flaps..  - simply everything where the value 0 does not help to represent a safe mode configuration). 

Right now there is no such option. It is possible to achieve the goal indirectly by accompanying the script with external mixes, logical switches and special functions, but it increases complexity. 

It would be an improvement and aid security if the 'failure output' (= output on dead script) could be specified right in the script:

Idea 1) Output declaration by specifying pairs of value names and failure output values:
_local output { {"ValA", -1024}, {"ValB", 0}, {"ValC", 140} }_
- (+) very explicit, no way to forget to specify the values
- (-) breaking change

Idea 2) Additional, optional failure defaults declaration:
_local output { "ValA", "ValB", "ValC"}_
_local failure_output { -1024, 0, 140}_
- (+) optional, old scripts still work
- (-) more room for errors, user needs to keep order of the elements in the two tables in sync

projectkk2glider, 2016-05-06:
Much simpler solution:
* use a fixed (or whatever you want) value that will be used in case of Lua script failure on the first mixer line
* use Lua script value on second line with **replace** modifier. 

When Lua script fails the first mixer line will be active.

Spainman, 2016-05-06:
I wasn't aware that the replace modifier has that behavior. That helps right now. Thank you!

Still it adds complexity and moves the definition of 'what happens when things go south' outside the script, requiring to take the precautions in every single model that uses the script. Also, if the script author != script user, the script user needs to understand how and why to add the required security logic himself. 

Mix scripts will always be a daunting task - in my personal opinion the easier it is to assure that functionality doesn't impact security, the better. Therefore I would appreciate if the enhancement would be considered.

kilrah, 2016-05-06:
Your request seems very limiting, as IMO in most use cases a fixed output in case of failure is not good enough. The system (a line coming from a mix script gets disabled if the script fails) was made so that you can have a default backup control.

Basically the practice that should be encouraged is to set up everything as if mix scripts didn't exist, then add the script on top if you want more complex / precise control so that you always have a "good enough" fallback.

Spainman, 2016-05-06:
Okay, thank you. I understand the idea now: A mix line where the source is lua mix output is completely bypassed if the corresponding script fails - irrespective of the multiplexing mode. That wasn't clear - the reference guide doesn't mention this important fact.  

That covers the case where the lua output is used right as the source of the line but I'm on a different use case and there are inconsistencies: 
**Output of a lua mix can be used as an argument of a logical switch**
- when the script is killed, the last output value seems to stay current (at least the evaluation result of the switch doesn't change)

**Use of such a logical switch inside as a source or switch of a mix line**
- consequently as the logical switch ignores that the lua mix died, the result of the mix line _stays_ although the script is dead

Please see the attached example [test.zip](https://github.com/opentx/opentx/files/252388/test.zip). It just demonstrates what I'm talking about. I haven't found out how to persist the selection of the mix script in the simulator or how to set it up in companion (is there any way?), so please select the contained mix.lua script manually.
- the script has a single constant output and can simply be killed with SH for testing
- all involved channels are controlled by the elevator stick and affected by the mix script
- CH1 - CH3 show the bypassing of the mix line when the script dies - great
- L1 evaluates the script output value and clearly results to true as long as the script runs (unfortunately also when the script is dead)
- CH4, Line2 uses L1 as source, when the script is dead it stays affected
- CH5, Line1 uses L1 as enabling switch, when the script is dead it stays affected

projectkk2glider, 2016-05-06:
> That wasn't clear - the reference guide doesn't mention this important fact. 

Would you be wiling to help improve the guide? The guide source is here https://github.com/opentx/lua-reference-guide

For the second part:

Firstly I think that your idea of defining *failsafe* values for the Lua script in the script itself is very inflexible. How would you change the failsafe value on the flying field without a computer?

Regarding your concerns about logical switches: you can always go trough a mixer (with properly configured *failsafe* value) and use mixer output in logical switches instead of direct Lua script output value. 

So using Lua scripts and planing for the script failure takes some thinking and planing, but I think it can already be implemented quite easy.

projectkk2glider, 2016-05-06:
I will also check how the logical switches are supposed to work in presence of Lua script failure...

projectkk2glider, 2016-05-06:
When Lua mixer script fails:
 * script outputs remain unchanged - the last value that was generated by the Lua script is kept. This (last) value is then used elsewhere (logical switches, inputs, mixes, etc..)
 * the only special handling is for mixer lines **where Lua script output is used as a source**: in such case the **whole mixer** line is skipped (behaves the same as that line is not present)
 * there is no special handling anywhere else - so the last value that was put out by the script before it was killed is used until the model is reloaded.

Spainman, 2016-05-07:
>Would you be wiling to help improve the guide? 

Sure! Actually I switched from using mercurial to git some days ago and was also educating myself on the github workflow in order to be able to contribute to OpenTx Speaker. Regarding the actual writing: Any tips on md editors with WYSIWYG on Win? I'm an ms-word person when it comes to writing text...

>Firstly I think that your idea of defining failsafe values for the Lua script in the script itself is very 
inflexible

You call it Inflexibility, to me it's encapsulation :-). Think of complex arming scenarios modeled as  finite state machines. Right now there is no way to encapsulate the whole fsm within the script because there is no way to define the transition to the 'dead-script'-state. 
But read on... I don't ask to replace known behavior, just augmentation of functionality.

> - the only special handling is for mixer lines where Lua script output is used as a source: in such 
case the whole mixer line is skipped (behaves the same as that line is not present)

I'm all for keeping the handling for the mix lines! Initially it wasn't clear to me how it works. It should of course stay. My idea would be to 
- add the functionality of the in-script defined fail-safes as an optional tool 
- allow to define the fail-safe outputs individually and allow to indicate that an output has no defined fail safe, e.g. by using nil
```
local output { "ValA", "ValB", "ValC"}
local failure_output { -1024, nil, 140} -- ValB has no failsafe
```
- if **no in-script fail safe** is defined for a specific output, then **apply the current behavior** of discarding the mix line when the script dies 
- if there is an **in-script fail safe** defined for the output, then **keep the line alive and use the specified fail safe value**

>there is no special handling anywhere else - so the last value that was put out by the script before it was killed is used until the model is reloaded.

This 'anywhere else' you mentioned is where the in-script failsafes would be a huge help. Having to pass every single output through a synthetic mixer just to assign a fail safe value, even when the output is never used as an actual mix line source, is a huge pita, error prone and complex to set up.

Also, if there is no reason why not to do so, I would suggest to allow the selection of script outputs as a source in inputs.

Thank you for taking your time to think about it and also  for looking up how things exactly work right now. I will try to document it in the guide. If you decide to implement the in-script failsafes, I would of course be willing to document that too.

projectkk2glider, 2016-05-07:
>Any tips on md editors with WYSIWYG on Win?

I will prepare as short wiki page with instructions on how to edit/contribute to the Lua docs. There is also dedicated chat room for Lua docs https://gitter.im/opentx/lua-reference-guide

As for your suggestion, I think we need to think this over some more.

closed 2020-04-06 by stale[bot]