opentx/opentx #3473: Inconsistent new model global variables initialisation by Companion and radio

elecpower, 2016-05-08:
The model created on the radio intialises the linked field for each GVar in FM 1-8 to "Flight mode 0 value" whereas Companion initialises to "Own value".

(Taranis X9D, Companion 2.1.8, Bootloader 2.1.7, Eeprom 217, Ubuntu 14.04, Windows 10)

projectkk2glider, 2016-05-08:
We need to decide which one is correct - radio or Companion and adapt other one to do the same.

elecpower, 2016-05-08:
Both default trims to "Use Trim from Flight Mode 0" so it would make sense from consistency to do the same for Gvars. 
There is some logic in FM0 being the default for all values and the starting point for all other FMs and you tweak each FM for its purpose.
If Companion is to change to the radio default the fix for #3214 will need to be revisited.

On a related point there is an an interface inconsistency in Companion when fields are linked to other FMs. For GVars the value field is disabled forcing you to go to the referenced FM to make a change (I agree with)  whereas trims are fully enabled and changes are pushed back to the referenced FM which has an element of risk especially if the default is FM0 (this is what happened in the first iteration of FM clear). The user may change the trim values and then set the linked field to Own Value not realising the changes have also been made to the previously linked FM. I would favour disabling the trims when linked to avoid the risk.

projectkk2glider, 2016-05-09:
I agree with @elecpower on both points the gvar initialization and trim disabling.

projectkk2glider, 2016-06-09:
@elecpower commit ce68f364255b4d878fe90581ce57872b122135fc needs to be ported to the `next` branch

elecpower, 2016-06-17:
master and next fixed so closing

PR #3578 (2016-06-08): Fixed #3473: New model global variables and rotary encoders initialisation and trim controls behaviour
Fixes #3473 Companion initialise global variables and rotary encoders to linked to Flight mode 0.
Companion trims disabled when linked to other flight mode except when + own as offset.
Radio initialise rotary encoders linked to Flight mode 0.
commit ce68f364 (2016-06-08): Fixed #3473: New model global variables and rotary encoders initialisation and trim controls behaviour (#3578)

* Fixes #3473: Inconsistent new model global variables

* Correct linked drop down index where not linked

* Modify flightModeData::clear to accept parameter
Modify radio defaults to set rotary encoders to linked to flight mode zero
closed 2016-06-17 by elecpower