opentx/opentx #3474: Companion UI bug: Mix line drag'n drop before CH1

Spainman, 2016-05-08:
Reproduction steps:
- edit model
- enter Mixes tab
- shift-drag a line of CH1 before CH01

       -> line appears before CH01
- shift-drag a line that does not belong to CH1 and is the only line in the corresponding channel

       -> line 'donor' channel moves completely before CH1

-> leaving and re-entering the mixes tab reestablishes a valid view
-> simulator radio menus show valid entries too when pressing Simulate from within the scrambled Mixes tab
==>> Perception-wise just an UI irregularity
Companion 2.1.8, Windows

projectkk2glider, 2016-05-08:
>shift-drag a line of CH1 before CH01

I can not do this on Linux

But yes, the shift-drag to any position is broken.

elecpower, 2016-06-18:
I can replicate on Ubuntu 14.04

commit 825adca0 (2016-09-22): Fixes #3474 Mix line drag drop before CH1 (#3838)
closed 2016-09-22 by kilrah