opentx/opentx #3475: [option ppmus] Keep logical values in Mixer Monitor

Spainman, 2016-05-08:
I just came across a situation where I needed to set up exact channel values ppm-wise to match the requirements of some custom hardware on the receiver side. So I flashed the firmware with ppmus checked. While the display in the Channel Monitor (= display of what goes to the receiver and where ppm actually matters) is exactly what I was looking for, the Mixer Monitor became pretty much useless because the values are also shown in ppm 'units'.

I doubt that there is a use case where seeing the intermediate, logical values (= mixer output) as pulse widths makes more sense than logical units. All the inputs and mixer parameters are configured using logical units (-100, 100), so reasoning about the correctness of the mixer outputs in logical units is natural.

I would appreciate if ppmus would
- continue to display ppm in the channel monitor and the outputs page (no change)
- but would not affect the mixer monitor (change, display logical units)

If you think that ppm in the mixer monitor actually is required, maybe there could be
- an option to switch the displayed units from within the mixer monitor on-the-fly (the menu button could be used)
- or a firmware switch to exclude ppm from the mixer monitor (e.g. ppmuslight)

projectkk2glider, 2016-05-09:
Well the simplest change would be to fix the Mixers monitor to percent. I am OK with that.

Second option (switch unit with MENU key) is also easily doable, but probably only for ARM based boards or perhaps only for Taranis (and Horus).

bsongis, 2016-05-10:
Yes the second option is nice!

Spainman, 2016-05-10:
So if it is the Menu key, I think it would great if the same switching between units would also be possible in the Channel Monitor (mainly as a conformance thing).

closed 2020-07-08 by stale[bot]