opentx/opentx #3476: [Edit Mix screen] Range indicator widget does not take into account weight sign

Spainman, 2016-05-08:
Title + screenshots below pretty much describe the whole problem

[image]
- positive weight -> indicated range correct

[image]
- negative weight -> indicated range inverted

tomkfly, 2016-05-09:
Surely that is correct. Range stays the same but stick/ switch works in the opposite direction.

kilrah, 2016-05-09:
+1, the point of this thing is to show the movement range, the direction is irrelevant.

Spainman, 2016-05-09:
I wasn't implying that the two numbers alone are the wrong way around, but the whole widget. Taking your comments into account, I assumed information embedded in the representation that isn't there. 

Let's leave sticks and switches aside and thing in terms of the resulting x/y-graph, that obviously isn't visible in the mix line screen.
Let's ignore what may have been specified for the widget behavior in the past and assume that
-  the widget serves as a **condensed representation of the graph**

Wouldn't the visualization as illustrated in the following
- make **sense**,
- transport **more useful information** than representing both graphs with the same widget,
generally **help to visualize the graph mentally** much better than representing both graphs with the same widget ?

[image]

tomkfly, 2016-05-09:
To me It's the same range,but you are travelling in the opposite direction, that makes more sense.

projectkk2glider, 2016-05-09:
I went ahead and did this, @kilrah or @bsongis you decide if  you like it :wink: #3476

mhotar, 2016-05-10:
I do not think it is right to merge. When the graph is swapped it will not match the one in the channel monitor. The range does not change with negative weight, from the channel point of view. We can add some graphics to the input line instead, to reliably indicete input is inverted.

Spainman, 2016-05-10:
I have never seen the widget as a static representation of the channel monitor indicators. Anyway - I brought it up so it's clear that I prefer functionality and a representation that is intuitively understandable over conformance with something that is on an entirely different screen.

projectkk2glider, 2016-05-12:
Reopening until we port the #3476 to the `next` branch.

PR #3477 (2016-05-12): Fixes #3476: better representation of negative mixer weight in offset…
… bar widget

Closes #3476 

It was an easy change, other devs can decide if we merge this.
commit 32f68047 (2016-05-12): Merge pull request #3477 from opentx/projectkk2glider/issue_3476_mixer_offset_bar

Fixes #3476: better representation of negative mixer weight in offset…
commit 494a83db (2016-05-12): Fixes #3476: better representation of negative mixer weight in offset bar widget
commit 49b3e542 (2016-05-12): Fixes #3476: better representation of negative mixer weight in offset bar widget (ported from master)
closed 2016-05-12 by projectkk2glider