opentx/opentx #3478: New native Curve Type similar to 'Single Expo' doable?

Spainman, 2016-05-09:
I find myself jumping through hoops whenever I try to add expo to controls that don't have a logical or mechanical center (throttle, flaps, wheel brakes). 

Would it be feasible to add a **new native curve type** in the **inputs** and **mixes** editors, that implements the same thing as the **"single expo"** curve generator in companion? See screenshot, coefficient would of course be the only required parameter - just like the normal symmetrical expo.

[image]

It would really be so incredibly useful - specially when tuning things at the field or even in the air using a pot and gvar.

Spainman, 2016-05-09:
On a second thought a **more flexible solution** would be to 
- add a **two numeric parameters** to the **current expo**, that allow to **set** the **inflection point (x, y)** 

This would for example allow to smooth the pitch input around the hover point in a rotorcraft really easily.

fraternl, 2016-05-11:
A similar (not the same) request has been made here: https://github.com/opentx/opentx/issues/1654

Futaba has a single expo selected as standard for throttle. The "normal" expo is not appropriate for throttle.

I prefer the less flexible "single expo", but I just fly planes.... 
Maybe 2 extra expo's?

closed 2020-04-06 by stale[bot]