opentx/opentx #3479: Feature request  Split Telemetry in 2 pages

fraternl, 2016-05-10:
The telemetry page is rather long and I think it would be more comfortable to have these split in 2

Page switching is faster than scrolling and the telemetry screens have not much to do with setting up the sensors.
The 1st screen could be called "sensor setup"

kilrah, 2016-05-10:
Just press the UP key to go to screen setup quicker.

fraternl, 2016-05-10:
It's not just speed.
IMHO it looks better

kilrah, 2016-05-10:
The screens could be moved to their own page indeed, that would harmonize with the Horus too. Not a fan of an extra page to go through in the model menus though.

bsongis, 2016-05-10:
Yes I do agree, these screens are not only used for telemetry now, they should move elsewhere. But where? A new entry in the contextual menu of the main views?

kilrah, 2016-05-10:
On the Taranis it makes less sense than on Horus given there are still standard screens... an extra menu page would maybe still be better

elecpower, 2016-05-10:
+1 for split.
Please consider those new to OpenTX and RC in any redesign. IMO paging is more natural to lead one to the topic area eg Mixes, Sensors, Telemetry. Burying access makes the learning curve just that much more frustrating especially when trying to make changes in the field.

projectkk2glider, 2016-05-11:
Extra page is my favourite.

schwabe, 2016-05-25:
I think 9056e764606263d28ec090674197053d47beed9a implemented this already and we can close this.

closed 2016-05-25 by schwabe