opentx/opentx #3480: Change discovery of sensors

fraternl, 2016-05-10:
When there was a problem with the discovery of not existing sensors I proposed to make the time for discovery limited during setup. This solved the problem of these sensors and now the discovery is not done all the time.

It was then implemented in a way that one needed to explicitly enable "discover sensors" and "stop discovery". I think that this is superfluous, clutters the screen unnecessary and gives some problems with beginners.

I want to propose that this "switch" is done implicitly by being on the telemetry page.
This would not need user interaction and also takes care of forgetting it to switch off.

schwabe, 2016-05-10:
no. Implicitly starting that is wrong. I often delete some sensors that I want to ignore (e.g. A2). Implicit autodiscovery would always add them back.

fraternl, 2016-05-10:
That feature was never intended for that.
If it wasn't buggy it would have always been on.

schwabe, 2016-05-10:
@fraternl I know that. I am a dev too. "Wrong" was probably the wrong word. I am just wanted to point out that I and maybe other are now (ab)using the feature for different purposes.

kilrah, 2016-05-10:
For me 2 solutions. Either the original issue has been fixed (possible, since some issues have been found and fixed with the serial port reads) and we remove the whole thing altogether (having sensors discovered automatically is the most user friendly), or it hasn't and we leave it like it is, the "sensor inhibit" function that was gained as a side effect being a useful feature (that I use too).

projectkk2glider, 2016-05-10:
I vote for no change.

opentx-jenkins, 2016-05-10:
Here I also use the possibility of inhibit some sensors like A2

projectkk2glider, 2016-05-27:
I think we agree about the no change, closing...

closed 2016-05-27 by projectkk2glider